compas-dev/compas #264: On Windows, COMPAS functions that rely on Numpy/Scipy complain about not finding certain DLLs

brgcode, 2019-03-07:
Running files in Sublime Text or in VS Code using the Debugger, complain about not finding certain Numpy/Scipy DLLs.

brgcode, 2019-03-07:
This seems related to the version of Numpy/Scipy. To resolve do
```
conda remove --force numpy scipy
pip install numpy scipy
```
This will install the most recent versions and then the problem goes away...

gonzalocasas, 2019-03-07:
mmmm... I don't think it's a very good idea to use `pip` versions of numpy/scipy when in conda. 
Which are those dlls that it cannot find?

brgcode, 2019-03-07:
basically all of them. it just fails at the first one (which tends to be the one underlying multiarray)

brgcode, 2019-03-07:
perhaps we can force conda to install specific version of numpy and scipy, but at the moment i find it very frustrating that a lot of conda stuff simply doesn't work properly. there are plenty of problems on mac with matplotlib and on windows with the DLLs...

brgcode, 2019-03-07:
everything is always fixable but it requires a lot of debugging and extreme careful tuning which seems very fragile...

gonzalocasas, 2019-03-07:
But installing numpy/scipy with pip on conda basically defeats the purpose of conda to begin with. I have seen issues with Pillow DLLs recently, but they are unrelated to numpy, is that the ones you mean?

gonzalocasas, 2019-03-07:
and yes, we should pin a specific version (or at least major.minor) of numpy.

brgcode, 2019-03-07:
no has nothing to do with pillow but those indeed also sometimes pop up

brgcode, 2019-03-07:
we should just find a stable distribution and stick with it for now...

gonzalocasas, 2019-03-08:
> we should just find a stable distribution and stick with it for now...
yep, agree, but more than simply just stick with it, what we need to do is to CI test the whole thing.

Could you please indicate a couple of the function calls that tend to break? I will setup tests on them (if you have tests, all the better, but if not, I'll write some) and an appveyor integration to run CI on Windows platform.

brgcode, 2019-03-12:
it seems that the DLL problem goes away starting with `numpy>=1.12`

brgcode, 2019-03-14:
will close this because it will be addressed by pinning down versions of `numpy`, `scipy`, `matplotlib`, ... for different versions of Python and for different platforms...

closed 2019-03-14 by brgcode