compas-dev/compas #274: def get_points from layer

mariklad, 2019-03-28:
method in compas_rhino.utilities get_points(var: layer) only works for None layer.

error returned: Message: LayerVisible() got an unexpected keyword argument 'force_visible'

Looking at the scr file replacing var: force_visible with forcevisible_or_donotpersist seems to solve the issue:
[image]

mariklad, 2019-03-28:
same for all get_lines, curves etc

brgcode, 2019-03-28:
which version of Rhino is this?

mariklad, 2019-03-28:
Rhino 6 SR12

brgcode, 2019-03-31:
it might be something that was changed between versions. will have a look at the differences and add a fix to the next (patch) release...

closed 2020-01-13 by brgcode