compas-dev/compas #276: Abort trap when instantiating MeshPlotter()

arpastrana, 2019-04-02:
**Describe the bug**
Instantiating `MeshPlotter()` terminates with uncaught exception of type NSException

**To Reproduce**
```
from compas.datastructures import Mesh
from compas.plotters import MeshPlotter

mesh = Mesh.from_obj(compas.get('faces.obj'))
plotter = MeshPlotter(mesh)
```

**Expected behavior**
Have the plotter open and displaying the loaded mesh.

**Desktop (please complete the following information):**
-  Compas Version: Fresh Installation of 0.5.1
 - OS: Mojave
 - Python version: 3.7
 - Python package manager: Anaconda

**Additional context**
Full exception print-out below.

```
2019-04-02 16:10:39.181 python[2619:857913] -[NSApplication _setup:]: unrecognized selector sent to instance 0x7f8c389244b0
2019-04-02 16:10:39.183 python[2619:857913] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x7f8c389244b0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff33adbecd __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff5fba3720 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff33b59275 -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x00007fff33a7db40 ___forwarding___ + 1486
	4   CoreFoundation                      0x00007fff33a7d4e8 _CF_forwarding_prep_0 + 120
	5   libtk8.6.dylib                      0x000000011c566154 TkpInit + 324
	6   libtk8.6.dylib                      0x000000011c4be0ee Initialize + 2622
	7   _tkinter.cpython-37m-darwin.so      0x0000000118bf3a3f _tkinter_create + 1183
	8   python                              0x000000010a706fe6 _PyMethodDef_RawFastCallKeywords + 230
	9   python                              0x000000010a8438b2 call_function + 306
	10  python                              0x000000010a841565 _PyEval_EvalFrameDefault + 46165
	11  python                              0x000000010a834eee _PyEval_EvalCodeWithName + 414
	12  python                              0x000000010a705ca7 _PyFunction_FastCallDict + 231
	13  python                              0x000000010a788321 slot_tp_init + 193
	14  python                              0x000000010a7922e1 type_call + 241
	15  python                              0x000000010a70699f _PyObject_FastCallKeywords + 175
	16  python                              0x000000010a843945 call_function + 453
	17  python                              0x000000010a841659 _PyEval_EvalFrameDefault + 46409
	18  python                              0x000000010a706795 function_code_fastcall + 117
	19  python                              0x000000010a843837 call_function + 183
	20  python                              0x000000010a841565 _PyEval_EvalFrameDefault + 46165
	21  python                              0x000000010a834eee _PyEval_EvalCodeWithName + 414
	22  python                              0x000000010a705ca7 _PyFunction_FastCallDict + 231
	23  python                              0x000000010a709ba2 method_call + 130
	24  python                              0x000000010a707622 PyObject_Call + 130
	25  python                              0x000000010a841760 _PyEval_EvalFrameDefault + 46672
	26  python                              0x000000010a834eee _PyEval_EvalCodeWithName + 414
	27  python                              0x000000010a706eb3 _PyFunction_FastCallKeywords + 195
	28  python                              0x000000010a843837 call_function + 183
	29  python                              0x000000010a841659 _PyEval_EvalFrameDefault + 46409
	30  python                              0x000000010a834eee _PyEval_EvalCodeWithName + 414
	31  python                              0x000000010a706eb3 _PyFunction_FastCallKeywords + 195
	32  python                              0x000000010a843837 call_function + 183
	33  python                              0x000000010a841659 _PyEval_EvalFrameDefault + 46409
	34  python                              0x000000010a706795 function_code_fastcall + 117
	35  python                              0x000000010a713400 property_descr_get + 96
	36  python                              0x000000010a765cba _PyObject_GenericGetAttrWithDict + 250
	37  python                              0x000000010a83ef62 _PyEval_EvalFrameDefault + 36434
	38  python                              0x000000010a706795 function_code_fastcall + 117
	39  python                              0x000000010a713400 property_descr_get + 96
	40  python                              0x000000010a765cba _PyObject_GenericGetAttrWithDict + 250
	41  python                              0x000000010a83ef62 _PyEval_EvalFrameDefault + 36434
	42  python                              0x000000010a706795 function_code_fastcall + 117
	43  python                              0x000000010a708e28 object_vacall + 248
	44  python                              0x000000010a709230 PyObject_CallFunctionObjArgs + 144
	45  python                              0x000000010a713616 property_descr_set + 38
	46  python                              0x000000010a76690b _PyObject_GenericSetAttrWithDict + 219
	47  python                              0x000000010a765720 PyObject_SetAttr + 96
	48  python                              0x000000010a83d8b3 _PyEval_EvalFrameDefault + 30627
	49  python                              0x000000010a834eee _PyEval_EvalCodeWithName + 414
	50  python                              0x000000010a705ca7 _PyFunction_FastCallDict + 231
	51  python                              0x000000010a788321 slot_tp_init + 193
	52  python                              0x000000010a7922e1 type_call + 241
	53  python                              0x000000010a70699f _PyObject_FastCallKeywords + 175
	54  python                              0x000000010a843945 call_function + 453
	55  python                              0x000000010a8415a0 _PyEval_EvalFrameDefault + 46224
	56  python                              0x000000010a834eee _PyEval_EvalCodeWithName + 414
	57  python                              0x000000010a898430 PyRun_FileExFlags + 256
	58  python                              0x000000010a8978a7 PyRun_SimpleFileExFlags + 391
	59  python                              0x000000010a8c5c17 pymain_main + 10775
	60  python                              0x000000010a6d942d main + 125
	61  libdyld.dylib                       0x00007fff60c71ed9 start + 1
	62  ???                                 0x0000000000000002 0x0 + 2
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

brgcode, 2019-04-03:
this is not really a bug, in the sense that it is a [known problem](https://matplotlib.org/faq/osx_framework.html) of the Anaconda Python install on Mac :) 

after setting up a virtual environment, make sure to also install `python.app`. this will make the Python installation of the environment a *Framework* installation, which basically adds `pythonw` to the available executables.

```bash
$ conda create -n myenv python=3.7 COMPAS
$ conda activate myenv
$ conda install python.app
```
in addition to that, you have to make sure that all calls involving `matplotlib` are handled by `pythonw` instead of `python`.

thanks to @gonzalocasas, if you are working with the Conda builder in Sublime Text, you can now add this is as a configuration preference:

```
Sublime Text > Preferences > Package Settings > Conda > Settings - User
```

```python
{
    "use_pythonw" : true
}
```

closed 2019-04-03 by brgcode

arpastrana, 2019-04-03:
Thanks, @brgcode! Though not explicitly an bug in `compas`, it would come in quite handy to include this nice (and already comprehensive) explanation in the documentation section. New (mac) users will appreciate it! 😄