compas-dev/compas #279: Online documentation finds matches but returns nothing

Achillx, 2019-04-15:
**Describe the bug**
Searching in the online documentation it finds matches but does not display them.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to compas-dev.github.io
2. Search for `cross product`
3. You(I) get `Search finished, found 8 page(s) matching the search query.`
4. But nothing is being displayed

**Expected behavior**
Display 8 pages that match the search query!

**Screenshoot**
[image]

**Desktop (please complete the following information):**
 - OS: [macOS 10.14.3 (18D42)]
 - Chrome Version 73.0.3683.103 (Official Build) (64-bit)

Achillx, 2019-04-18:
Sometimes it returns fewer that what was found and it might return duplicates.

[image]

brgcode, 2019-04-26:
i do some filtering of the results after they are found. was a bit of a quick fix because the search function returns stuff that it shouldn't, but seems to result in the above behavior.

will be fixed when i roll out the new website...

brgcode, 2019-07-19:
better now?

closed 2019-08-14 by brgcode