compas-dev/compas #285: importing `Viewer` exits Python

inesariza, 2019-05-24:
when importing `Viewer` nothing happens. it seems to exit Python without even printing `mesh`. No error message given.

```
from compas.datastructures import Mesh
from compas.viewers import Viewer

mesh = Mesh.from_polyhedron(6)
print(mesh)
viewer = Viewer()
print(viewer)
```

[image]

Issue found on Windows, Python 3.6.7, compas 0.6.1 installed via conda

romanarust, 2019-05-27:
I don't know why python crashes, but you're probably missing 'PyOpenGL' and 'PySide2', which are optional  installs with compas.
Try installing with `conda install pyside2`.
`conda install pyopengl` did not work for me, I downloaded the whl from
https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyopengl and installed with `pip install PyOpenGL-3.1.3b2-cp36-cp36m-win_amd64.whl`

inesariza, 2019-05-27:
update: both packages are installed via conda with no errors.
but still same result: python crashes when executing `from compas.viewers import Viewer`

GeneKao, 2019-05-27:
@inesariza try to install pyqt as well and see if it works
```
conda install -c anaconda pyqt 
```
If not can you conda list your packages? my pyopengl is from conda. 

```
pyopengl                  3.1.3b2                    py_1    conda-forge
pyparsing                 2.3.0                      py_0    conda-forge
pyqt                      5.6.0            py36h655552a_6    anaconda
pyside2                   5.6.0a1         py36hc43cc3a_1005    conda-forge
pytest                    4.4.0                    pypi_0    pypi
python                    3.6.7             h8dc6b48_1004    conda-forge
```

inesariza, 2019-05-28:
now it works! problem is solved with pip install `vtk` package from the .whl instead from conda:

sequence of events:
I installed `pyside2`, `pyopengl`, and `pyqt` via conda with no errors. 
`from compas.viewers import Viewer` > python crashes
`import vtk` > python crashes
`pip install VTK-8.1.2-cp36-cp36m-win_amd64.whl` > Requirement already satisfied
`pip install --force  VTK-8.1.2-cp36-cp36m-win_amd64.whl` > Successfully installed VTK-8.1.2
`from compas.viewers import Viewer` > works and the viewer shows up! 🎈 

I get another error when running the viewer but will post that one in separate issue.

thanks @romanarust @GeneKao @gonzalocasas

closed 2019-05-28 by inesariza

gonzalocasas, 2019-05-28:
I was fairly sure `vtk` was installed as part of our `compas` [conda recipe](https://github.com/conda-forge/compas-feedstock/blob/master/recipe/meta.yaml), but it is *not*. I vaguely remember having lots of trouble with it back last year, and it seems we left it as an optional dependency. I will test a bit to see if we can add it.