compas-dev/compas #286: Wrong closest_point_in_cloud documentation

Achillx, 2019-05-24:
**Describe the bug**
The documentation says that it returns the distance to the closest point, but instead it returns the squared distance!
`closest_point_in_cloud` uses `sort_points` which uses `distance_point_point_sqrd`

**Expected behavior**
Write that it returns the squared distance, or return `closest_point_in_cloud` to return `sqrt(data[0])`.

yck011522, 2019-06-13:
Yes, I agree it is probably fine for the sorting function to use squared distance to sort. But if these distances are to be returned, perhaps it is better to properly document it. Otherwise everyone expects a clostestpoint function to return normal distance.

I'm not sure if other function that uses 'distance_point_point_sqrd' are also incorrectly documented.

brgcode, 2019-06-16:
thanks for the report.
will fix in next release...

brgcode, 2020-04-14:
solved since 92cbf6f8

closed 2020-04-14 by brgcode