compas-dev/compas #287: AttributeError: module 'clr' has no attribute 'AddReferenceToFileAndPath'

feihln, 2019-06-05:
**Describe the bug**
When importing Network from compas.datastructures, I get an error in xalglib.py .
_AttributeError: module 'clr' has no attribute 'AddReferenceToFileAndPath'_

**To Reproduce**
Steps to reproduce the behavior:
1. Context : 
originally in Pycharm2019.1 but same error in VSCode1.34.0 .
conda environment with only compas as an extra package.
python: 3.7.3
compas: 0.6.2 

2. Sample script:
`import compas`
`from compas.datastructures import Network`

3. Sample data: None

4. See error:
```
Traceback (most recent call last):
  File "C:/Users/nico/PycharmProjects/LearningCompas/Networks_test1.py", line 2, in <module>
    from compas.datastructures import Network
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\datastructures\__init__.py", line 143, in <module>
    from .network import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\datastructures\network\__init__.py", line 6, in <module>
    from ._network import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\datastructures\network\_network.py", line 17, in <module>
    from compas.geometry import centroid_points
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\geometry\__init__.py", line 419, in <module>
    from .bestfit import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\geometry\bestfit\__init__.py", line 7, in <module>
    from .bestfit_numpy import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\geometry\bestfit\bestfit_numpy.py", line 11, in <module>
    from compas.numerical import pca_numpy
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\__init__.py", line 118, in <module>
    from .fd import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\fd\__init__.py", line 6, in <module>
    from .fd_alglib import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\fd\fd_alglib.py", line 12, in <module>
    from compas.numerical._alglib._core import xalglib
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\_alglib\__init__.py", line 5, in <module>
    from ._core import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\_alglib\_core\__init__.py", line 5, in <module>
    from ._array import *
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\_alglib\_core\_array.py", line 12, in <module>
    from compas.numerical._alglib._core import xalglib
  File "C:\Users\nico\Anaconda3\envs\compas_env\lib\site-packages\compas\numerical\_alglib\_core\xalglib.py", line 47, in <module>
    clr.AddReferenceToFileAndPath(_candidate)
AttributeError: module 'clr' has no attribute 'AddReferenceToFileAndPath'
```

**Expected behavior**
well... no error.

**Screenshots**
no screenshots.

**Desktop (please complete the following information):**
 - OS: Windows10, version 1809
 - Python version: 3.7.3
 - Python package manager : conda 

**Additional context**

clr version is 2.4.0
it seems it does not have AddReferenceToFileAndPath method...
dir(clr) returns: 
`['AddReference', 'FindAssembly', 'GetClrType', 'ListAssemblies', 'Microsoft', 'Python', 'System', '_AtExit', '__class__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', '__version__', '_extras', 'clrModule', 'clrmethod', 'clrproperty', 'e__NativeCall', 'getPreload', 'setPreload']`
the environment is fresh and with only compas added. 

the same code works fine in IronPython (Rhino).

This is the first time I submit a bug report, so I hope I did it correctly.
Nico

brgcode, 2019-06-16:
hi nico,

thanks for the report. it looks great :)
will look in to this as soon as possible.

tom

gonzalocasas, 2019-06-17:
@brgcode I believe the package that @feihln has installed is not that `clr`, but `pythonnet`: https://pypi.org/project/pythonnet/ which effectively, implements parts of the `clr` module, but not the `AddReferenceToFileAndPath` method.

```
(test-env) $ python
Python 3.6.7 (default, Feb 28 2019, 07:28:18) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import clr
>>> dir(clr)
['AddReference', 'FindAssembly', 'GetClrType', 'ListAssemblies', 'Microsoft', 'Python', 'System', '_AtExit', '__class__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', '__version__', '_extras', 'clrModule', 'clrmethod', 'clrproperty', 'e__NativeCall', 'getPreload', 'setPreload']
>>>
```

commit 18ef3d70 (2019-06-19): fix #287: quick hack to resolve the unsupported pythonnet `clr` module
commit 79641c9e (2019-06-19): Merge pull request #293 from compas-dev/quick-fix-clr-pythonnet

fix #287: quick hack to resolve the unsupported pythonnet `clr` module
commit 44202d9e (2019-07-19): Merge branch 'master' into inherit_attributes

* master:
  removed unnecessary code
  Update changelog
  fix #287: quick hack to resolve the unsupported pythonnet `clr` module
  Updated CHANGELOG.md
  Added face_neighborhood to Mesh class
  rereference internal compas_plotters imports
closed 2019-06-19 by brgcode