compas-dev/compas #295: [Docs] cross referencing in documentation?

yijiangh, 2019-06-21:
**Describe the bug**
How can I do cross-referencing in the rst files for the documentation, following the [Sphinx doc on rst](http://www.sphinx-doc.org/en/stable/usage/restructuredtext/roles.html#cross-referencing-documents)?
(All experiments performed on Windows, python 3.7.3, compas version 0.6.2)

**To Reproduce**
For example, I placed a label in the `gettingstarted.rst`:
```
.. _getting-started--virtual-environments:
Virtual environments
====================
```

And I want to reference it in the `contribution.rst`:
```
.. _getting-started--virtual-environments:
Virtual environments
====================
```

By `invoke docs`, the generated html will look like the following, i.e. plain text without any link:
[image]

I'm wondering if I'm doing anything wrong or this is something related to the compas doc template?

brgcode, 2019-06-24:
can't think of a reason why this would have anything to do with the COMPAS theme, but i will check.

in any case, there is a typo in your example (two hyphens between started and virtual in the label and only one in the reference). if that also exists in the code then that will be the reason for the referencing to fail...

brgcode, 2019-07-19:
is this now resolved?

yijiangh, 2019-08-02:
Sorry for the delayed reply... Yes I tested it today it works - it doesn't matter if you are using `-` or `--` or `_` in your naming.

A side note, for developers, I think it's better to use `sphinx-build -b html docs dist/docs`  (without the [`-E`](https://www.sphinx-doc.org/en/master/man/sphinx-build.html) option as specified by the `docs` function in the `tasks.py` file) so that sphinx-build "only read and parse source files that are new or have changed since the last run".

Rebuilding the whole docs really takes forever if you are doing write-and-build iterations...

closed 2019-08-02 by yijiangh