compas-dev/compas #297: [compas_rhino] import compas error in Rhino under non-admin Windows account

yijiangh, 2019-06-23:
**Describe the bug**

I'm using compas with a **non-admin account** in Windows 10.  `compas` and `compas_rhino` are installed successfully in Anaconda prompt **with admin privilege**. But `compas` cannot be imported in Rhino 6.0.

**To Reproduce**

In Windows 10, I am using a non-admin account. The admin installed Rhino 6.0 for all users. Thus, both users have their own `C:\Users\<user name>\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)` directory. I have the password for the admin account, but I am trying to install and use compas in my own non-admin account.

I can install `compas_rhino` successfully by running the Anaconda prompt with admin privilege:
[image]

After restarting Rhino 6, I can't import `compas` in Rhino's python editor.
[image]

**Potential cause for the bug**

The snapshot below shows the different results by running `whoami` in the Anaconda prompt with admin privilege (upper) and without (lower), under my non-admin Windows account.
[image]

Installing `compas_rhino` requires admin privilege,  which will make the symlink set to the admin's Rhino IronPython but not the in the non-admin account's.

yijiangh, 2019-06-23:
Commit https://github.com/compas-dev/compas/commit/f334785f92346ba9b630121acc3d485c6f7776c5 fixes this bug.

With https://github.com/compas-dev/compas/commit/f334785f92346ba9b630121acc3d485c6f7776c5, I can run Anaconda prompt **without admin privilege**, run `python -m compas_rhino.install` without any problem, and find `compas` successfully installed for Rhino IronPython:

[image]

The only annoyance is that it will pop the privilege escalation dialog multiple times (once per package), which is acceptable for me at this moment.

@gonzalocasas @brgcode : With this, I will consider this bug fixed, but I will keep it open until commit f334785f92346ba9b630121acc3d485c6f7776c5 is merged.

brgcode, 2019-06-25:
@gonzalocasas could this also be the root of the problems with RPC on some computers?

brgcode, 2019-06-25:
@gonzalocasas the non-admin solution throws errors on RhinoMac. will document properly later today...

gonzalocasas, 2019-06-25:
> the non-admin solution throws errors on RhinoMac. will document properly later today...

@brgcode could you post a stack trace or similar? I thought the change should not affect rhinomac at all, this only happens (or should happen) on the windows code path.

gonzalocasas, 2019-06-25:
> @gonzalocasas could this also be the root of the problems with RPC on some computers?

I would not think of admin rights as the cause of RPC problems immediately, because RPC uses a non-privileged port, but on the hand, maybe there is something else that is failing on non-admin accounts during the rpc server start up.

brgcode, 2019-06-25:
```
Message: libdl.so

Traceback:
  line 353, in __init__, "/Applications/Rhinoceros.app/Contents/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib/ctypes/__init__.py"
  line 441, in <module>, "/Applications/Rhinoceros.app/Contents/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib/ctypes/__init__.py"
  line 6, in <module>, "/Applications/Rhinoceros.app/Contents/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib/compas/_os.py"
  line 30, in <module>, "/Applications/Rhinoceros.app/Contents/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib/compas/__init__.py"
  line 7, in <module>, "/Users/vanmelet/Code/BlockResearchGroup/compas_hilo/scripts/fabric/02_assign-strips.py"
```

gonzalocasas, 2019-06-25:
ah ok! I'll fix that

commit d0f91cbe (2019-06-26): Merge pull request #299 from compas-dev/non-admin-install

Fixes #297: Support install to Rhino with non-admin accounts
PR #832 (2021-05-11): Fix GH userobjects symlink
Fixes #297
When the native symlinking does not work, we fallback to a polyfill implementation that was creating the symlinks as directory links on Windows.

### What type of change is this?

- [x] Bug fix in a **backwards-compatible** manner.
- [ ] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these out after creating the PR. If you're unsure about any of them, [... cut]
closed 2019-06-26 by brgcode