compas-dev/compas #311: meshes_weld precision in units rather string format

Achillx, 2019-08-02:
# Feature Request

As a user, I want to input precision in project units rather in string format so that I can control how close the vertices should be to get welded.

brgcode, 2019-08-06:
you mean something like
```python
compas.PRECISION = 0.001
```
instead of
```python
compas.PRECISION = '3f'
```

Achillx, 2019-08-06:
Yes, but it doesn't have to be link to `compas.PRECISION`. It can have its own precision, if that makes it easier.

robin-oval, 2019-08-06:
and also allow something like
`compas.PRECISION = 10`

brgcode, 2019-08-06:
no the problem is that it is more difficult to use precision in that format in the context of geometric maps

brgcode, 2019-08-06:
anyway, i will try to see if i can implement something like this...

https://stackoverflow.com/questions/6189956/easy-way-of-finding-decimal-places

brgcode, 2019-08-06:
> and also allow something like
> `compas.PRECISION = 10`

would that be `10` digits?

robin-oval, 2019-08-06:
no, more the rounding precision: 11 becoming 10

brgcode, 2019-08-07:
added a function for setting precision. for now it only handles floating point precision. for example

```python
compas.set_precision(0.0001)
# compas.PRECISION => '4f'
```

closed 2019-08-07 by brgcode