compas-dev/compas #312: welding a mesh, slightly shifts its final geometry

Achillx, 2019-08-02:
**Describe the bug**
When welding a mesh, its vertices' coordinates get rounded based on the precision value. This results in a slightly shifted mesh. This is linked to `reverse_geometric_key` function.

**Expected behavior**
Only the welded vertices should be moved.

brgcode, 2019-08-06:
makes sense.
will have a look...

robin-oval, 2019-08-06:
I did the same in the meshes_join_and_weld function

brgcode, 2019-08-06:
@robin-oval 

> I did the same in the meshes_join_and_weld function

what?
make sure that the points don't move or use `reverse_geometric_key`?

robin-oval, 2019-08-06:
I due to how I use the `reverse_geometric_key` the points move in both functions. They're all snapped to the precision.

brgcode, 2019-08-06:
okay but it is easy enough to fix, you just need to keep track of the original point coordinates in a dict. will try something later today.

Achillx, 2019-08-06:
> I did the same in the meshes_join_and_weld function

`meshes_join` do not use `geometric_key`, only the welding functions

brgcode, 2019-08-06:
can you give me some sample data or a snippet to test this on?

brgcode, 2019-08-06:
btw, should `meshes_join_and_weld` not just be a combination of `mesh_join` and `meshes_weld`?

robin-oval, 2019-08-06:
yes

brgcode, 2019-08-06:
i pushed a new version that i tested with the following snippet

```python
from compas.datastructures import Mesh
from compas.datastructures import mesh_weld
from compas_plotters import MeshPlotter

vertices = [[0, 0, 0], [0.04, 0, 0], [1.0, 0, 0], [1.0, 1.0, 0], [0, 1.0, 0]]
faces = [[0, 1, 2, 3, 4]]

mesh = Mesh.from_vertices_and_faces(vertices, faces)
mesh = mesh_weld(mesh, precision='1f')

plotter = MeshPlotter(mesh, figsize=(10, 7))
plotter.draw_vertices(text={key: "{:.3f}".format(mesh.vertex[key]['x']) for key in mesh.vertices()}, radius=0.03)
plotter.draw_edges()
plotter.show()
```

brgcode, 2019-08-06:
the only thing is that you have no control over which vertex stays in place, even though you might have opinions about this :)

closed 2019-08-07 by brgcode