compas-dev/compas #313: Unable to install compas with Blender python

solub, 2019-08-04:
Hi, (thank you for providing a tool like compas in Blender)

When trying to install compas with `pip `command from python in the Blender folder, I get the following error:

```
C:\Program Files\Blender Foundation\Blender\2.80\python\bin>python.exe -m pip install compas --user
Collecting compas
  Using cached https://files.pythonhosted.org/packages/1b/47/42a015826e4312e230db4a7d5f44ccdd353a67ddecd813ee3a166a208d0c/COMPAS-0.7.1-py2.py3-none-any.whl
Requirement already satisfied: matplotlib>=2.2; python_version >= "3.5" in c:\users\solub\appdata\roaming\python\python37\site-packages (from compas) (3.1.1)
Collecting planarity (from compas)
  Using cached https://files.pythonhosted.org/packages/91/d6/33cc8784b375a45505326e28cc01ed987e29a8091aa7c70de0c10aee7704/planarity-0.4.1.tar.gz
Requirement already satisfied: sympy in c:\users\solub\appdata\roaming\python\python37\site-packages (from
compas) (1.4)
Requirement already satisfied: numba in c:\users\solub\appdata\roaming\python\python37\site-packages (from
compas) (0.45.1)
Requirement already satisfied: cython in c:\users\solub\appdata\roaming\python\python37\site-packages (from compas) (0.29.13)
Collecting numpy==1.15.4 (from compas)
  Using cached https://files.pythonhosted.org/packages/00/0e/5a8c34adb97fc1cd6636d78050e575945e874c8516d501421d5a0f377a6c/numpy-1.15.4-cp37-none-win_amd64.whl
Requirement already satisfied: scipy>=1.0 in c:\users\solub\appdata\roaming\python\python37\site-packages (from compas) (1.3.0)
Requirement already satisfied: networkx in c:\users\solub\appdata\roaming\python\python37\site-packages (from compas) (2.3)
Requirement already satisfied: pillow in c:\users\solub\appdata\roaming\python\python37\site-packages (from compas) (6.1.0)
Requirement already satisfied: kiwisolver>=1.0.1 in c:\users\solub\appdata\roaming\python\python37\site-packages (from matplotlib>=2.2; python_version >= "3.5"->compas) (1.1.0)
Requirement already satisfied: cycler>=0.10 in c:\users\solub\appdata\roaming\python\python37\site-packages (from matplotlib>=2.2; python_version >= "3.5"->compas) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in c:\users\solub\appdata\roaming\python\python37\site-packages (from matplotlib>=2.2; python_version >= "3.5"->compas) (2.4.2)
Requirement already satisfied: python-dateutil>=2.1 in c:\users\solub\appdata\roaming\python\python37\site-packages (from matplotlib>=2.2; python_version >= "3.5"->compas) (2.8.0)
Requirement already satisfied: setuptools in c:\users\solub\appdata\roaming\python\python37\site-packages (from planarity->compas) (39.0.1)
Requirement already satisfied: mpmath>=0.19 in c:\users\solub\appdata\roaming\python\python37\site-packages (from sympy->compas) (1.1.0)
Requirement already satisfied: llvmlite>=0.29.0dev0 in c:\users\solub\appdata\roaming\python\python37\site-packages (from numba->compas) (0.29.0)
Requirement already satisfied: decorator>=4.3.0 in c:\users\solub\appdata\roaming\python\python37\site-packages (from networkx->compas) (4.4.0)
Requirement already satisfied: six in c:\users\solub\appdata\roaming\python\python37\site-packages (from cycler>=0.10->matplotlib>=2.2; python_version >= "3.5"->compas) (1.12.0)
Installing collected packages: planarity, numpy, compas
  Running setup.py install for planarity ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\Program Files\Blender Foundation\Blender\2.80\python\bin\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\solub\\AppData\\Local\\Temp\\pip-install-ww1kgojx\\planarity\\setup.py'"'"'; __file__='"'"'C:\\Users\\solub\\AppData\\Local\\Temp\\pip-install-ww1kgojx\\planarity\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\solub\AppData\Local\Temp\pip-record-37060ik9\install-record.txt' --single-version-externally-managed --compile --user
--prefix=
         cwd: C:\Users\solub\AppData\Local\Temp\pip-install-ww1kgojx\planarity\
    Complete output (27 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.7
    creating build\lib.win-amd64-3.7\planarity
    copying planarity\planarity_functions.py -> build\lib.win-amd64-3.7\planarity
    copying planarity\planarity_networkx.py -> build\lib.win-amd64-3.7\planarity
    copying planarity\__init__.py -> build\lib.win-amd64-3.7\planarity
    creating build\lib.win-amd64-3.7\planarity\tests
    copying planarity\tests\test.py -> build\lib.win-amd64-3.7\planarity\tests
    copying planarity\tests\test_planarity.py -> build\lib.win-amd64-3.7\planarity\tests
    copying planarity\tests\test_planarity_networkx.py -> build\lib.win-amd64-3.7\planarity\tests
    copying planarity\tests\__init__.py -> build\lib.win-amd64-3.7\planarity\tests
    running build_ext
    cythoning planarity/planarity.pyx to planarity\planarity.c
    C:\Users\solub\AppData\Roaming\Python\Python37\site-packages\Cython\Compiler\Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: C:\Users\solub\AppData\Local\Temp\pip-install-ww1kgojx\planarity\planarity\planarity.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    building 'planarity.planarity' extension
    creating build\temp.win-amd64-3.7
    creating build\temp.win-amd64-3.7\Release
    creating build\temp.win-amd64-3.7\Release\planarity
    creating build\temp.win-amd64-3.7\Release\planarity\src
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.21.27702\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MT -Iplanarity/src/ "-IC:\Program Files\Blender Foundation\Blender\2.80\python\include" "-IC:\Program Files\Blender Foundation\Blender\2.80\python\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.21.27702\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /Tcplanarity\planarity.c /Fobuild\temp.win-amd64-3.7\Release\planarity\planarity.obj
    planarity.c
    planarity\planarity.c(4): fatal error C1083: Impossible d'ouvrir le fichier includeÿ: 'Python.h'ÿ: No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.21.27702\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Program Files\Blender Foundation\Blender\2.80\python\bin\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\solub\\AppData\\Local\\Temp\\pip-install-ww1kgojx\\planarity\\setup.py'"'"'; __file__='"'"'C:\\Users\\solub\\AppData\\Local\\Temp\\pip-install-ww1kgojx\\planarity\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\solub\AppData\Local\Temp\pip-record-37060ik9\install-record.txt' --single-version-externally-managed --compile --user --prefix= Check the logs for full command output.
```

Another I've noticed is that whenever I try to install a module with:

`C:\Program Files\Blender Foundation\Blender\2.80\python\bin>python.exe -m pip install modulename --user`

... that module doesn't appear in the `site-packages` folder of the Blender's Python.
I'm suspecting a conflict between the difference Python installations on my system but have no clue how to fix this.

Would you mind help me figuring out this issue ?

Regards,

gonzalocasas, 2019-08-04:
@solub the way to go to get compas correctly installed into blender is to use Anaconda, check out the docs here: https://compas-dev.github.io/main/gettingstarted/cad/blender.html

solub, 2019-08-04:
@gonzalocasas  Thank you for the swift reply. 

I did try to use Anaconda but when I type:

`ren python original_python`

I get an `access denied` message and can't proceed further in the installation.

solub, 2019-08-05:
Spent the whole day on this, I would really appreciate some help.
Copying/pasting the compas folders directely in `C:\Program Files\Blender Foundation\Blender\2.80\python\lib\site-packages\` doesn't work either... this thing is a nightmare.

brgcode, 2019-08-05:
hi,

i wrote a few things about this in response to your question on the forum.
in any case, don't try hacking away by copy-pasting stuff or using `pip` in combination with the Blender Python executable because that will just not work.

the goal of the procedure is to replace the Python used by Blender with the Python you installed in a conda environment that also contains COMPAS. so the instructions in the docs only make sense if that works.

1. where did you install Blender?
2. are you using the Anaconda Prompt to issue the instructions?
3. is the environment you installed COMPAS in active while you follow the instructions?
4. what other versions of Python do you have installed on your system?

closed 2019-08-13 by brgcode