compas-dev/compas #320: Proxy() raises exception when starting server in Rhino Mac 6

arpastrana, 2019-09-06:
**Describe the bug**

While launching a `Proxy()` server in rhino mac 6.0, a `ValueError` exception is raised by rhino's `subprocess.py` stating: _"Cannot redirect stderr to stdout yet."_.

Having a closer look at it, this issue arises when instantiating `Popen()`in proxy's `start_server()` method as the system complains about the arguments being passed to the `stdout` and `stderr` parameters of the constructor. In its current state, it looks as follows:

`self._process = Popen(args, stdout=PIPE, stderr=STDOUT, *args, **kwargs)`

Went to check whether the issue would persist should an `XFunc()` would be called, and to discard if this was a _shady rhino-for-mac-update-issue_, but to my surprise this operated fine.

Therefore, after inspecting what is dissimilar between `Proxy()` and `XFunc()`, I have got around the problem by setting `stderr` equal to `PIPE`; looking in sum as follows:

`self._process = Popen(args, stdout=PIPE, stderr=PIPE, *args, **kwargs)`

Nevertheless, I am not sure if this modification would have any potential side effects down the road.

**To Reproduce**
```
from compas.rpc import Proxy
from compas.utilities import XFunc

my_xfunc = XFunc('myfavoriteexternalmethod')
print('xfunc worked like a charm')

my_proxy = Proxy()
print('proxy worked like a charm too')
```

**Expected behavior**
`Proxy()` server should be started smoothly.

**Desktop:**
 - OS: OSX Mojave
 - Python version: IronPython 2.7.5 in Rhino Mac 6.0
 - Python package manager. Anaconda

arpastrana, 2019-09-06:
(If my quick fix is alright, I could post a PR 😄)

brgcode, 2019-09-07:
there is something wrong with the handling of `stderr` and `stdout` in the context of the Proxy servers in any case, peraps we should not even redirect at all...

proposing a pull request is a good idea, then the issue gets added to our todo list :)

arpastrana, 2019-09-09:
PR created as #321.

yijiangh, 2019-10-24:
Just for documentation purposes, this PR ("by coincident") fixed the `Exception: Fault` error message when users are trying to use `RPC`  and `Proxy` with [VRay for Rhino](https://www.chaosgroup.com/vray/rhino) installed on their computer. I tested it on my Windows 10 laptop, python `3.7`, VRay for Rhino `4.00.02`. 

This bug was reported when using `RPC` and `Proxy` in `compas 0.7.2` in a workshop with @stefanaparascho.