compas-dev/compas #324: calculate rotation between two frames

xarthurx, 2019-09-16:
**Describe the bug**
When calculating translation/rotation between two given frames `F1` and `F2` using the method `from_frame_to_frame`, the transformation matrix is different if the two frames are moved, but relative position/orientation is maintained.

**To Reproduce**
```
F1 = Frame([0.0, 0.0, 0.0], [1.0, 0.0, 0.0], [0.0, 1.0, 0.0])
F2 = Frame([0.0, 0.0, 0.0], [1.0, -1.0, 0.0], [1.0, 1.0, 0.0])

trans1 = Transformation.from_frame_to_frame(F1, F2)

F3 = Frame([1.0, 1.0, 0.0], [1.0, 0.0, 0.0], [0.0, 1.0, 0.0])
F4 = Frame([1.0, 1.0, 0.0], [1.0, -1.0, 0.0], [1.0, 1.0, 0.0])
trans2 = Transformation.from_frame_to_frame(F3, F4)

R1 = Rotation.from_frame_to_frame(F1, F2)
R2 = Rotation.from_frame_to_frame(F3, F4)

print R2
```
Output:

```
[[    0.7071,    0.7071,    0.0000,   -0.4142],
 [   -0.7071,    0.7071,    0.0000,    1.0000],
 [    0.0000,    0.0000,    1.0000,    0.0000],
 [    0.0000,    0.0000,    0.0000,    1.0000]]
```

**Expected behavior**
```
[[    0.7071,    0.7071,    0.0000,    0.0000],
 [   -0.7071,    0.7071,    0.0000,    0.0000],
 [    0.0000,    0.0000,    1.0000,    0.0000],
 [    0.0000,    0.0000,    0.0000,    1.0000]]
```
For transformation, I'm not sure. But for rotation between to frames that have the same point, Quaterian matrix should not contain a translation `w` vector?

Why rotating around the shared axis of the two frames will result in a translation?

It seems this is the design of the method that `worldXY` is used as an intermediate frame to compute the two, but in this case, the location of the two frames in the world coordinates will matter... 
This will inhibit the application, say, calculate the transformation between two given frame, and apply the same relationship to a 3rd frame to get the 4th, so that `1vs2` and `3vs4` have the same transformation.

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Python version [e.g. 2.7]
 - Python package manager [conda]

romanarust, 2019-09-17:
Thanks for pointing out on the `Rotation.from_frame_to_frame`: it simply inherited the method from `Transformation`, so not clearing translational values. This should be fixed now.

However, the behavior of `Transformation.from_frame_to_frame` is correct, it returns the same as `Rhino.Geometry.Transform.PlaneToPlane`. You can check with:
```
from compas_ghpython import draw_frame
import Rhino.Geometry as rg

f1 = draw_frame(F1)
f2 = draw_frame(F2)
f3 = draw_frame(F3)
f4 = draw_frame(F4)

print(rg.Transform.PlaneToPlane(f1, f2))  # same values as trans1
print(rg.Transform.PlaneToPlane(f3, f4))  # same values as trans2
```

xarthurx, 2019-09-17:
Thanks for the quick response.

For the behavior, I agree that it is not a bug.

However, the current design of the system seems missing sth for mapping a transformation from one object to the other, as all transform happens in the global system, as well as the `Frame.from_transformation`.

In the current setup, in order to achieve a transformation mapping, one has do to multiple "global->local->global" frame change to get there, which is not very convenient.

romanarust, 2019-09-17:
Could you please add an enhancement request with a concrete example of how you want to operate that indicates what features you want to have implemented.

commit 4616a61e (2019-09-17): fix #324