compas-dev/compas #326: AttributeError: 'Mesh' object has no attribute 'neighbors'

Licini, 2019-09-17:
**To Reproduce**
Steps to reproduce the behavior:
```
>>> import compas
>>> from compas.datastructures import Mesh
>>> mesh = Mesh.from_obj(compas.get('faces.obj'))
>>> mesh.vertex_neighborhood_centroid(0)
```

**Expected behavior**
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/lichen7/WORK/compas/src/compas/datastructures/mesh/_mesh.py", line 2579, in vertex_neighborhood_centroid
    return centroid_points([self.vertex_coordinates(nbr) for nbr in self.neighbors(key)])
AttributeError: 'Mesh' object has no attribute 'neighbors'
>>> 

```

 - OS: OS X
 - Python version 3.7
 - Python package manager conda

brgcode, 2019-09-17:
should be `self.vertex_neighbors` i guess

Licini, 2019-09-18:
right, that make senses, gonna fix it in next pull

closed 2019-09-21 by brgcode