compas-dev/compas #335: Transform multiple frames

romanarust, 2019-09-19:
# Feature Request

Similar to `transform_points` and `transform_vectors` it would be great to have `transform_frames`.
Likewise, `Frame` should be extended with `represent_(frames\points\vectors)_in_(local\global)_coordinate_system`

xarthurx, 2019-09-19:
the `transform_frames` can be simple made by transforming three basis and one point of the given frame?

gonzalocasas, 2019-09-20:
As a side comment, I think we need to rename `represent_(frames\points\vectors)_in_(local\global)_coordinate_system` to a significantly more compact name?

I was checking how other libraries and tools do this, and I like the way [matlab](https://in.mathworks.com/help/phased/ref/local2globalcoord.html) does it, so I would suggest the following:
1. Replace all `represent_(frames\points\vectors)_in_(local\global)_coordinate_system` methods with two **static methods**: `local_to_global(obj, local)` and `global_to_local(obj, local)`.
1. Add `local_to_global(self, local)` and `global_to_local(self, local)` as instance methods to `Point`, `Vector` and potentially other classes as well. 
1. The new static methods of `Frame` determine the kind of argument as follows: if it's a list of float, it's assumed to represent a point. Anything else needs to be passed as an instance (of `Vector`, `Frame`, etc).
1. If the argument is an instance, it needs to support the interface defined in item 2 (`local_to_global(self, local)` and `global_to_local(self, local)`), so the supported types is not hard-coded.
1. An additional, functional API could optionally be created as well, but I would question its value, because the static method proposed here basically serves the same function.

Once that is in place, the usage of this API would look like this:
```
# Defining some local coordinate system
rcs = Frame((100, 150, 0), (1, 0, 0), (0, 1, 0))

point = (10, 1, 10)
point_l = Frame.global_to_local(point, local=rcs)
point_g = Frame.local_to_global(pg, local=rcs)
```

romanarust, 2019-09-20:
@gonzalocasas: Good suggestion!
I'll soon start a branch working on this, there are some other tasks related to this:
https://github.com/compas-dev/compas/projects/23#card-26510777

PR #357 (2019-10-18): Update `Frame`, `Transformation`
This PR adds several features and suggests some little modifications. Fixes #335.
Checkout Changelog for a complete list of additions and changes.

Comes with 2 incompatible API changes:
* To streamline with the rest, changed parameters `origin, uvw` of `global_coords_numpy` and `local_coords_numpy` to `frame`
* Renamed `Frame.represent_point/vector/frame_in_global_coordinates` and `Frame.represent_point/vector/frame_in_local_coordinates` to `Frame.local_coords` and `Frame.global_coords` 
 
### What type of change is this?

- [ ] Bug fix in a **backwards-compat [... cut]
closed 2019-10-18 by brgcode