compas-dev/compas #336: Broken link to compas_viewers and compas_plotters in documentation

gonzalocasas, 2019-09-19:
The documentation page listing [packages](https://compas-dev.github.io/packages.html) includes `compas_viewers` and `compas_plotters` but both of them return 404s when trying to open them.

brgcode, 2019-09-21:
link to `compas_plotters` is working now.
`compas_viewers` is more complicated because travis can't import the package (dependencies PySide and PyOpenGL are not installed...)

brgcode, 2019-09-21:
btw should we also rename `compas_plotters` to `compas_mpl` and `compas_viewers` to `compas_opengl` to make things more explicit? (we also don't have a `compas_cad` and `compas_cadfordotnet` :)

gonzalocasas, 2019-09-21:
> btw should we also rename `compas_plotters` to `compas_mpl` and `compas_viewers` to `compas_opengl` to make things more explicit? (we also don't have a `compas_cad` and `compas_cadfordotnet` :)

No, I don't think so. As a `compas_viewer` user, you don't really care what's the underlying rendering technology, you just want to get the standalone viewport functionality. The direction of the integration is only one-way, it's outwards from `compas`. 

The CAD integrations are completely different, you are the user of a specific CAD, e.g. Rhino, and want to integrate `compas` into it, so you do care about the specific technology. In this case, the integration is bi-directional.

brgcode, 2019-09-21:
okay, i will add a manually written doc page for now, so we can close the issue :)

closed 2019-09-22 by brgcode