compas-dev/compas #341: Documentation tied to version of compas pkg.

tetov, 2019-09-29:
# Documentation tied to version of compas pkg.

## Problem description

I'd guess that the majority of users are installing compas from pypi or conda which means that they are installing the repo at a [tagged commit](https://github.com/compas-dev/compas/tags). If they want to view the [documentation online](https://compas-dev.github.io/main/) however they will see documentation from the most recent commit, which could cause confusion.

So the correct (version specific) documentation will be available in the installed package but not online.

## Proposed solution

I think it would be good to consider a system that allows documentation tied to tagged commits.

Two examples with both works with Sphinx would be:
* [sphinxcontrib-versioning](https://github.com/sphinx-contrib/sphinxcontrib-versioning)
* [ReadTheDocs](https://readthedocs.org/)

Sphinxcontrib-versioning could probably work with our current `Invoke` solution while a solution like ReadTheDocs would mean having the docs built and hosted by another service.

If there's interest I'd happily help making this change.

brgcode, 2019-09-29:
hi anton,

yes, that is indeed an unsolved problem that causes confusion. especially when we make backwards incompatible changes to the API.

as far as i know, it is not possible to use your own theme with RTD, so i would prefer the `sphinxcontrib-versioning` solution.

would be helpful indeed if you could propose a solution.

thanks,
tom

tetov, 2019-09-29:
Hi Tom!

I see. Yeah, RTD only allows its own theme or the standard Sphinx theme.

I will try to get the versioning-extension to work in the coming week. There's is some questions regarding the maintenance of the project, see <https://github.com/sphinx-contrib/sphinxcontrib-versioning/issues/59>, but there seems to be people interested and keen to keep it going.

tetov, 2019-09-30:
[image]

Easiest would be to add the version selection like mockup above. Would that be a good solution?

brgcode, 2019-09-30:
if clicking on the version number switches the docs to the corresponding version, then that is great :)

worbit, 2019-10-01:
pasting here from a parallel discussion:

> the “problem” is indeed related to this. however, in the specific case of delaunay and trimesh, the api docs do correspond to the current conda package version (0.7.2) but the examples and tutorials reflect an older state. not sure how these get generated…

**context:** we encountered in the MAS compas introduction, that the example [mesh-delaunay-from-points](https://compas-dev.github.io/main/examples/rhino/mesh-delaunay-from-points.html) and [mesh-remeshing](https://compas-dev.github.io/main/examples/rhino/mesh-remeshing.html) both import from `compas.topology` intsead of `compas.geometry` and `compas.datastructures` respectively.

tetov, 2019-10-01:
> pasting here from a parallel discussion:
>
>
> **context:** we encountered in the MAS compas introduction, that the example [...]

I'm currently updating the examples (`docs/examples/`) to make them work with v0.7.2. I'm not sure the examples as they are set up now could be tied to versions.. I'll have a look.

brgcode, 2019-10-01:
the examples will need some manual maintenance. there is no way around this. we should clearly mention that people have to report broken examples on the issue tracker...

tetov, 2019-10-01:
Examples updated, see PR https://github.com/compas-dev/compas/pull/343
Found a tutorial that needs updating but might be better for someone else to look at since I don't really understand the tutorial yet haha. See issue https://github.com/compas-dev/compas/issues/344

tetov, 2019-11-19:
Documentation will reflect latest release, PR #396. Closing.

tetov, 2019-11-27:
Adding this as a note to future self, we would need to amend old tagged commits to add  sphinxcontrib_versioning to the conf.py extension array.
https://github.com/sphinx-contrib/sphinxcontrib-versioning/issues/70#issuecomment-559024887

tetov, 2020-07-15:
This might be something to look at again, I've been following the development of a new sphinx plugin, [sphinx-multiversion](https://holzhaus.github.io/sphinx-multiversion/master/index.html).

yck011522, 2020-09-23:
Generic Ping.

I see that `compas_fab` have this versioning already. I don't know the implementation details but maybe we can use the same system? https://github.com/compas-dev/compas_fab/pull/196

Maybe @gonzalocasas have some experience?

closed 2020-11-24 by brgcode