compas-dev/compas #345: Unused files in docs/examples/rhino

tetov, 2019-10-01:
Might be time to clean out clutter in [docs/examples/rhino](https://github.com/compas-dev/compas/tree/master/docs/examples/rhino) if they are not planned to be implemented.

These that are not referenced in RST-files (i.e. not used for the docs website):

* bbox_oriented_volume.py
* coons_srf.3dm
* drx_rhino.3dm
* drx_rhino.py
* fd_numpy-remote-call.py
* formfinding-cablenet-rhino.gif
* formfinding-cablenet-rhino-interactive.jpg
* formfinding-cablenet-rhino.jpg
* interface-matlab-rhino.3dm
* interface-matlab-rhino.jpg
* interface-matlab-rhino.py
* mesh-planarization.3dm
* mesh-planarization.jpg
* mesh-planarization.py
* mesh-skeleton-modeling.gif
* mesh-smoothing.3dm
* mesh-smoothing.png
* mesh-smoothing.py
* mesh-stanford-armadillo.jpg
* mesh-stanford-armadillo.py
* mesh-stanford-dragon.jpg
* mesh-stanford-dragon.py

tetov, 2019-10-01:
Fixed in PR #343.

tetov, 2019-10-01:
Closed wrong issue, really sorry.

brgcode, 2019-10-15:
python files should be kept (and their inputs as well). rest can probably go. but we need to figure out a way to make more and better examples...

closed 2020-11-24 by brgcode