compas-dev/compas #350: Rhino dev/debug mechanism

Licini, 2019-10-07:
When developing compas_rhino, each time after code updated, one has to quit Rhino completely and re-run compas installation for Rhino, and then re-open Rhino for the changes to take effect. 
Do we currently have any mechanism in place to help faster this process?

closed 2019-10-07 by Licini

brgcode, 2019-10-07:
the modules have to be reloaded, which you can do by restarting rhino or using the reload function in `compas_ghpython` (which should mybe be moved into `compas.utilities`). re-installing should never be necessary...

gonzalocasas, 2019-10-07:
The function is already part of `compas_rhino`, I don't know if it's justified to move it even higher to core. I'd err on the side of nope