compas-dev/compas #352: unit tests for compas_rhino

Licini, 2019-10-08:
It appears really tricky to run pytest in Rhino, should we:

- use pre-built library like unittest from python instead?
- Or just write some simple script to browse and execute all test functions under `tests/compas_rhino`?

gonzalocasas, 2019-10-10:
I don't think is possible to run unit tests for the front ends (Rhino or GHPython or Blender). At least not without doing horrible trickery. For the most part, I would not focus too much effort on unit testing these packages; and rather put emphasis on the core packages, those that contain the logic bits, not the visual ones.

gonzalocasas, 2019-10-10:
btw, using a different lib (e.g. `unittest`) won't really help much; because tests inside rhino cannot run in travis-ci