compas-dev/compas #359: is_line_line_colinear

yck011522, 2019-10-16:
# Feature Request

As a **COMPAS user**, I want **compas.geometry.queries to have a function for lines co-linear check** and so that **I can use it**.

## Details

**Describe the solution you'd like**
I would like the following (or something similar) to happen:
```
line1 = Line([0, 0, 0], [1, 1, 1])
line2 = Line([2, 2, 2], [3, 3,3])
assert is_line_line_colinear(line1, line2) == true
"Another Direction should work too"
line3 = Line([-2,- 2, -2], [3, 3,3])
assert is_line_line_colinear(line1, line3) == true
```

**Describe alternatives you've considered**
Currently there is a function to check if 3 points in colinear. 
https://github.com/compas-dev/compas/blob/16f3f6ef3b49a6a2191f41feb5da79d7bdac3f5c/src/compas/geometry/queries.py#L117

There is also feature to do intersection between 2 lines.  
https://github.com/compas-dev/compas/blob/16f3f6ef3b49a6a2191f41feb5da79d7bdac3f5c/src/compas/geometry/queries.py#L792

But not a simple 2 lines colinear check, or did I miss something? Would be useful to have this feature.

similarly, is_line_line_overlap(line,line) might also be useful

brgcode, 2019-10-17:
so something that would basically do this?
```python
def is_line_line_colinear(line1, line2):
    a, b = line1
    c, d = line2
    return is_colinear(a, b, c) and is_colinear(a, b, d)
```

brgcode, 2019-10-17:
not sure what you mean by "overlap".
but if my interpretation is correct then two lines that are colinear always overlap.
perhaps you want to find out if the two segments between the respective start and end points overlap?

yck011522, 2019-10-17:
For the `is_line_line_colinear(line1, line2)`, I think there needs a tolerance setting.

> so something that would basically do this?
> 
> ```python
> def is_line_line_colinear(line1, line2):
>     a, b = line1
>     c, d = line2
>     return is_colinear(a, b, c) and is_colinear(a, b, d)
> ```

I'm not sure how tolerance will be checked in this case if you use is_colinear(pt,pt,pt) as a check. I assume a meaningful tolerance in this case is an angle (some sort of epsilon?). So the signature would be like `is_line_line_colinear(line1, line2, epsilon=1e-6)`

------

For the `is_line_line_overlap(line1, line2, epsilon=1e-6)` what i mean is that, since a line in COMPAS is not an "infinite line", two lines that are colinear will have 4 possibilities:

(using above a,b,c,d notation) 

1. Non overlapping : neither c nor d , is located between ab
2. Partially overlapping : either c or d (one only) , is located between ab
3. Full overlapping : ( a == c and b == d ) or ( a == d and b == c ) 
4. Containment: (Both c and d , is located between ab) or (Both a and b , is located between cd)

brgcode, 2019-10-18:
> since a line in COMPAS is not an "infinite line"

that is not correct though. it is true that a line in COMPAS is defined by two points, but it IS infinite. intersections between lines or between lines and other objects are found anywhere on the lines and not just on the part between the two points.

in fact, there is a `intersection_segment_segment` to do exactly that...

one could, however, assume that when the question of overlap is asked it is about the segments between start and end points and not about the lines because this would not make too much sense :)

brgcode, 2019-10-18:
anyhow, since with this thread the colinear function is basically implemented, how about we go through the proper channels, which means you write a PR and i approve it :)

yck011522, 2019-10-18:
Hi Tom, I haven't got any experience creating a Pull Request. 
But I can give it a try. Please let me know if I do anything wrong.

commit 49ef8057 (2019-11-07): Merge pull request #387 from yck011522/feature/is_line_line_colinear

closes #359: Added is_line_line_colinear()
commit bc838af3 (2019-11-07): closes #359: Added is_line_line_colinear()
closed 2019-11-07 by brgcode