compas-dev/compas #360: Issue running test_mesh.py testing cube_ascii.stl

tetov, 2019-10-16:
**Describe the bug**
Started having a test fail in test_mesh.py fail, [line 74](https://github.com/compas-dev/compas/blob/4e12566473f7479d2ac99ec6d65fa44b7673f168/tests/compas/datastructures/test_mesh.py#L56), while working on PR #348. (4008 vertices instead of expected 4020)

[The mesh in question](https://github.com/compas-dev/compas/blob/master/data/cube_ascii.stl)

While trying to figure out what went wrong I discovered that fresh installs of both current COMPAS 0.8.1 as well as 0.7.2 reports `mesh.number_of_vertices()` to be 4008. This has been tested both in `virtualenv` and `conda` on more than one machine.

Blender agrees with `test_mesh.py` reporting 4020 vertices while Rhino reports the following:
**TLDR:** non manifold, intersections and 24048 vertices.
```
General information about this mesh:

Mesh has 660 non manifold edges.
Skipping face direction check because of positive non manifold edge count.
Mesh has 10709 pairs of faces that intersect each other.
  This can cause problems if you're doing mesh boolean operations with it.

Mesh does not have any degenerate faces.
Mesh does not have any ngons.
Mesh does not have any extremely short edges.
Mesh does not have any naked edges.
Mesh does not have any duplicate faces.
Mesh has 172 disjoint pieces.
Mesh does not have any unused vertices.

ID: 893f94b4-db11-4879-abee-e38cdb9a4af1 (728)
Object name: (not named)
Layer name: Layer 01
Render Material: 
  source = from layer
  index = -1

Geometry:
  Valid mesh.
  Closed double precision polygon mesh: 24048 vertices, 8016 faces with normals
    Bounding box: (-71.1734,-42.8274,99.5188) to (77.8611,63.5517,177.751)
```

The test runs and seems to have run fine on Travis for all this time

Has anyone got any ideas what's happening here?

tetov, 2019-10-17:
Paging @Licini since you wrote the tests :). Any thoughts on this specific mesh?

brgcode, 2019-10-17:
it is a precision thing.

```python
import compas
compas.set_precision(0.0000001)
mesh = Mesh.from_stl(compas.get('cube_ascii.stl'))
v = mesh.number_of_vertices()
print(v)
```

finds 4020...

brgcode, 2019-10-17:
perhaps the STL reader should determine the required precision for the parser by checking the number of digits in the X coordinate of the first vertex...

tetov, 2019-10-17:
Ok! Thanks!

I can add that for ascii in PR #348 and for binary in future PR.

tetov, 2019-10-18:
Closed due to @brgcode answer, and discussion regarding inferring precision moved to #362.

closed 2019-10-18 by tetov