compas-dev/compas #363: RobotModel from URDF does not apply link's rotation

matteo-pacher, 2019-10-18:
I use `robot = RobotModel.from_urdf_string(urdf)` of the following URDF (snippet):
```
<link name="link">
    <visual>
      <origin rpy="0 0 3.14159265359" xyz="0.0 0.0 0.0"/>
      <geometry>
        <mesh filename="package://rfl/meshes/visual/y_gantry.stl"/>
      </geometry>
      <material name="">
        <color rgba="0.9254902 0.9254902 0.9058824 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 3.14159265359" xyz="0.0 0.0 0.0"/>
      <geometry>
        <mesh filename="package://rfl/meshes/collision/y_gantry.stl"/>
      </geometry>
    </collision>
  </link>
```
and then use the RobotArtist to visualize.

The issue seems to be that the displayed **link** geometry (both visual and collision) is not rotated.

gonzalocasas, 2019-10-31:
Confirmed, there seems to be a problem, either with origin and/or axis when there's a rotation specified in the urdf itself.

closed 2020-01-22 by romanarust

matteo-pacher, 2020-01-22:
Thank you Romana!

commit 5daec677 (2020-01-22): checking for origin attr in visual or collision, closes #363