compas-dev/compas #368: Add coding guidelines to CONTRIBUTING.md

tetov, 2019-10-23:
# Feature Request

As a contributor, I want coding guidelines specified so that I know how to format my code and docs and so that project uses a consistent style.

## Details

Some of this is already specified in [.editorconfig](https://github.com/compas-dev/compas/blob/master/.editorconfig)

**Describe the solution you'd like**
Specify in [CONTRIBUTING.md](https://github.com/compas-dev/compas/blob/master/CONTRIBUTING.md) that the project uses numpy style docstrings.

Specify which formatter should be used, i.e. `autopep8` or my favorite (opinionated) formatter [`yapf`](https://pypi.org/project/yapf/).

tetov, 2019-10-23:
See also related issue #239 for discussion on api docs.

tetov, 2019-10-23:
Per IRL conversation with @brgcode some settings are found in [`setup.cfg`](https://github.com/compas-dev/compas/blob/master/setup.cfg):

```
max-line-length = 180
exclude = */migrations/*
```

gonzalocasas, 2019-10-25:
I would suggest to encode guidelines fully into automatic checks, and then, in `CONTRIBUTING.md` simply refer to the fact that `requirements-dev.txt` installs all these tools that automatically check for this. Besides the numpy doc style, most of the rest is already partially there; but it needs to be re-checked in a bit more concerted effort.

tetov, 2019-10-25:
Are you talking about addIng a task for running files through black/yapf to fix them, or linter output in invoke test? The latter would probably an irritant for all contributers..

gonzalocasas, 2019-10-25:
We have `invoke check` to do that already, but several checks are disabled atm. But I mean more like the linter/editorconfig stuff, that, if the correct tools are installed and the IDE supports them, then you get all the red indicators directly in the editor. VS Code should already be doing that, because we install `autopep8` and have all the settings in the various setting places + editorconfig for the whitespace wars ;)

tetov, 2019-10-25:
Ok, then we might need to add that to the CONTRIBUTING guidelines, saying that autopep8 with these settings should be used.

My vim setup uses system flake8 even if the project wants it a different way and I think that might not be too uncommon.

tetov, 2019-11-06:
Seems like @brgcode got this started :). <https://github.com/compas-dev/compas/wiki/Developer-Guide>

Is this replacing CONTRIBUTING.md or in addition to it? (ie, should I add a link to the existing file or just make it link to the wiki?)

jf---, 2024-04-06:
@tomvanmele keep hitting that satisfying close button 
it ain't green for nuthin'

closed 2024-04-08 by tomvanmele