compas-dev/compas #379: ambiguous msg when running compas_rhino.uninstall

inesariza, 2019-10-30:
`python -m compas_rhino.uninstall`

returns this ambiguous message:

```
Uninstalling COMPAS packages from Rhino 6.0 IronPython lib:
Packages installed from environment: ...
   compas               OK
   compas_ghpython      OK
   roslibpy             OK
   compas_fab           OK
   compas_rhino         OK
   compas_bootstrapper  OK

Completed.
```

gonzalocasas, 2019-10-30:
@inesariza would this be better?

```
Uninstalling COMPAS packages from Rhino 6.0 IronPython lib:
The following packages have been detected and will be uninstalled: ...
   compas               OK
   compas_ghpython      OK
   roslibpy             OK
   compas_fab           OK
   compas_rhino         OK
   compas_bootstrapper  OK

Uninstall completed.
```

inesariza, 2019-10-31:
perfect!

gonzalocasas, 2019-10-31:
@inesariza merci!

commit 95638460 (2019-10-31): fix #379: clarify uninstall console output
closed 2019-10-31 by gonzalocasas