compas-dev/compas #383: print(Quaternion) bug

matteo-pacher, 2019-11-05:
**Describe the bug**
`__repr__` method for Quaternion class not working properly in IronPython.

**To Reproduce**
Steps to reproduce the behavior:
1. Context: Grasshopper IronPython
2. Sample script:
```
from compas.geometry import Quaternion
q = Quaternion(1, 0, 0, 0)
print q
```
3. See error:
```
Runtime error (ArgumentTypeException): format() argument after * must be a sequence, not Object_1$1

Traceback:
  line 100, in __repr__, "C:\Users\pacherm\Documents\Repositories\compas\src\compas\geometry\primitives\quaternion.py"
  line 9, in script
```

**Expected behavior**
The method should print the Quaternion representation.

**Desktop**
 - OS: Windows
 - Python version: IronPython

**Additional context**
I tested the method both in python 2.7 and 3.7 and the method doesn't throw any error.

Licini, 2019-11-06:
thx for the ticket, I will be fixing this issue in a new PR

matteo-pacher, 2019-11-06:
Thank you for the fix!

closed 2019-11-06 by matteo-pacher