compas-dev/compas #388: Documentation style of multiple-optional-attribute-function very hard to understand.

yck011522, 2019-11-07:
# Feature Request

As a API user, I want easier-to-read documentation for functions where the input type / input structure is not obvious. 

## Details

I was trying to use the function compas_rhino.draw_lines.
I found only [placeholder documentation](https://compas-dev.github.io/main/api/generated/compas_rhino.utilities.draw_lines.html?highlight=draw_lines#compas_rhino.utilities.draw_lines) about the function, so I went to the [source code.](https://github.com/compas-dev/compas/blob/bd9b044961d8ce250043cb14860a741023d29489/src/compas_rhino/utilities/drawing.py#L193)
Perhaps I come from a typed language background, but the signature of the function ` def draw_lines(lines, **kwargs): ` does not tell me any information how to use the function. I find it very time-consuming to have to dig into the implementation of the code to know how I should format the input. 

**Describe the solution you'd like**
A clear and concise documentation should help but,
A better structured function signature will be much faster to understand. Perhaps optional variables can have default vales within the signature.

brgcode, 2019-11-11:
yup understood. will address as soon as possible...

yck011522, 2020-11-11:
I believe this is fixed.

closed 2020-11-11 by yck011522