compas-dev/compas #397: "module has not object BoxArtist" or cannot "import BoxArtist from compas_rhino.artists"

inesariza, 2019-11-17:
Cannot import BoxArtist

should it be included in __init__ file?
https://github.com/compas-dev/compas/blob/master/src/compas_rhino/artists/__init__.py

brgcode, 2019-11-18:
ideally, yes, but the `Shape` artists are not ready to be released yet. should have not been in the docs... did you find a reference to it somewhere else as well?

brgcode, 2019-11-18:
by the way, the reason for this is that it is not clear if it makes sense to have a `Shape` artist at all, since all shapes would have to be visualised using a mesh. therefore, they would just provide a very thin wrapper around the `MeshArtist`.

gonzalocasas, 2019-11-18:
I believe `Mesh.from_shape()` is included in the last release, so using the `MeshArtist` for this would be the way to go atm.

closed 2019-11-18 by brgcode