compas-dev/compas #400: LineArtist.draw() settings contains spelling error

maxenceryan, 2019-11-22:
**Describe the bug**
LineArtist settings for line color have a small spelling mistake.

On line 65 of lineartist.py, the key for self.settings should be color.line but is currently color.Line (the L is capitalized). Also see line 55.
This is giving me a key error when i try to use the draw() function from the line artist and inputting a line as input. 

Thanks :)

Licini, 2019-11-22:
Thanks! we are fixing this in new release

maxenceryan, 2019-11-24:
Great! feel free to close the issue once it is done.

brgcode, 2019-11-25:
the typo is fixed in the latest build. with an install from source it should therefore already be available. the next version that includes the fix will be released in the following days...

closed 2019-11-25 by brgcode