compas-dev/compas #452: Lazy initialization of readers and parsers of file classes

beverlylytle, 2020-01-09:
Within the collection classes from the files module, some of the doc strings promise the possibility not only of reading certain file types, but also writing them.  The current structure of these classes requires reading a file on initialization.  However, it seems that a user might want to first create some variation on a compas data structure, say a list of meshes, and then write that list to an obj file.  The suggestion of this issue is to still initialize the file classes OBJ/STL/etc with a file path and a precision, but to initialize the associated readers and parsers lazily or when a to-be-introduced read method is called on the OBJ/STL/etc. instance.  Such a change would work well with the current plans for a glTF exporter.

beverlylytle, 2020-01-09:
@tomvanmele What do you think?

brgcode, 2020-01-10:
agreed!

tetov, 2020-01-10:
2nd. Writing files will be moved to the file modules (probably with a new parent class; base_writer) as per discussions with @brgcode and @gonzalocasas, which means the file modules are not only used for reading.

brgcode, 2020-02-04:
can this be closed?

beverlylytle, 2020-02-04:
Yes.

closed 2020-02-04 by beverlylytle