compas-dev/compas #458: Convenience functions to convert from CAD geo obj < - > COMPAS geo obj

tetov, 2020-01-19:
I keep having to convert Rhino.Geometry objects to COMPAS and back when working in Grasshopper and ended up writing some functions to make it more convenient.

Would this be useful to include in `compas_rhino` or `compas_grasshopper`?

I discussed it a bit with @gonzalocasas and we were talking about having them in `compas_rhino.constructs` and `compas_rhino.artists`.

See gist for examples:
https://gist.github.com/tetov/6a4ddd44c9bdbc2422a45299e111a1b1

Suggestions of better function names would be good as well..

tetov, 2020-01-19:
`rg.Transform` <-> matrix as list of lists might also be useful. Added to the gist.

brgcode, 2020-01-21:
can you give an example of a ghpython script where this is used?

tetov, 2020-01-22:
Here are two examples from my current project.

The use case is when the input is Rhino.Geometry but I want to use a compas method, like `compas.geometry.axis_angle_vector_from_matrix` or when I want to visualize a frame from a `compas.robots` method as a plane in Rhino.

* https://github.com/tetov/rcf/blob/26ffb99bdd1672bed1a488e650d7b42ff58839f4/rcf/ur/ur_utils.py#L260

* https://github.com/tetov/rcf/blob/26ffb99bdd1672bed1a488e650d7b42ff58839f4/rcf/utils.py#L57

brgcode, 2020-01-23:
have a look at `compas_rhino.geometry`. especially `Point`, `Line`, `Plane`...
there should be examples in the `ifnameismain` section.

let me know what you think

tetov, 2020-01-23:
I took a quick look and that looks like what I'm looking for! And I guess some PRs for additional conversions wouldn't be bad? :)

brgcode, 2020-01-23:
that would not be bad, indeed. be aware of the difference between Rhino classes closely related to doc objects, and the structs that represent pure geometry...

brgcode, 2020-02-04:
can this be closed?

closed 2020-02-04 by brgcode

tetov, 2020-02-04:
Close it. I’ll open a pr for the ones I’m missing soon.