compas-dev/compas #479: compas relies on compas_hpc which isnt pip installable

jf---, 2020-02-14:
When building a new `conda` env, the following occurs

```python
IPython 3.7.6 (default, Jan  8 2020, 13:42:34) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.12.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import compas                                                                                               

In [2]: compas.__version__                                                                                          
Out[2]: '0.15.0'

In [3]: from compas.geometry import Frame                                                                           
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-3-d4e3751b2d47> in <module>
----> 1 from compas.geometry import Frame

~/miniconda3/envs/aec_control_room/lib/python3.7/site-packages/compas/geometry/__init__.py in <module>
    446 
    447 from .bbox import *  # noqa: F401 F403
--> 448 from .bestfit import *  # noqa: F401 F403
    449 from .hull import *  # noqa: F401 F403
    450 from .interpolation import *  # noqa: F401 F403

~/miniconda3/envs/aec_control_room/lib/python3.7/site-packages/compas/geometry/bestfit/__init__.py in <module>
      7 from .bestfit import *  # noqa: F401 F403
      8 if not compas.IPY:
----> 9     from .bestfit_numpy import *  # noqa: F401 F403
     10 
     11 

~/miniconda3/envs/aec_control_room/lib/python3.7/site-packages/compas/geometry/bestfit/bestfit_numpy.py in <module>
     16 from compas.geometry import local_to_world_coords_numpy
     17 
---> 18 from compas.numerical import pca_numpy
     19 
     20 

~/miniconda3/envs/aec_control_room/lib/python3.7/site-packages/compas/numerical/__init__.py in <module>
    104 from .ga import *  # noqa: F401 F403
    105 from .fd import *  # noqa: F401 F403
--> 106 from .drx import *  # noqa: F401 F403
    107 from .dr import *  # noqa: F401 F403
    108 from .devo import *  # noqa: F401 F403

~/miniconda3/envs/aec_control_room/lib/python3.7/site-packages/compas/numerical/drx/__init__.py in <module>
      9 
     10 if not compas.IPY:
---> 11     from .drx_numba import *  # noqa: F401 F403
     12 
     13 

~/miniconda3/envs/aec_control_room/lib/python3.7/site-packages/compas/numerical/drx/drx_numba.py in <module>
     22 from compas.numerical.drx.drx_numpy import _create_arrays
     23 
---> 24 from compas_hpc.geometry import cross_vectors_numba as cross
     25 from compas_hpc.geometry import dot_vectors_numba as dot
     26 from compas_hpc.geometry import length_vector_numba as length

ModuleNotFoundError: No module named 'compas_hpc'

```

brgcode, 2020-02-17:
hi jelle,

thanks for pointing this out.
could you check with version 0.15.1 and let me know if the problem is resolved?

thanks,
tom

jf---, 2020-02-17:
> could you check with version 0.15.1 and let me know if the problem is resolved?

Yes sir, that was fast, thx!

closed 2020-02-17 by jf---