compas-dev/compas #483: Add to_data() from_data() methods for RobotModel

yck011522, 2020-02-20:
# Feature Request

As a user, I create my `RobotModel` using code, with Mesh attached to Links and Joints defined.

But I currently cannot serialize this and therefore I cannot save and load it from disk. I expect methods such as `from_data()` and `to_data()` to exist, as consistent with other compas classes.

This will also allow deep copies of the `RobotModel`, and subsequently `RobotModel.copy()`.

brgcode, 2020-04-07:
@gonzalocasas could you have a look at this?

gonzalocasas, 2020-04-07:
I take it into our backlog. However, to be honest, right now this does not have very high prio, so, it will be a while until we get to it.

yck011522, 2020-04-07:
@gonzalocasas  Thanks for your honesty 😅

closed 2020-07-24 by beverlylytle