compas-dev/compas #495: volmesh_from_polysurface

salmamzfr, 2020-03-12:
**Describe the bug**
tried this:

> from compas_rhino.geometry._constructors import volmesh
> import rhinoscriptsyntax as rs
> guids = rs.GetObject(“select a closed polysurface”, filter=rs.filter.polysurface)
> volmesh = volmesh.volmesh_from_polysurfaces(VolMesh, guids)

returns:

> Message: iteration over non-sequence of type Guid
> 
> Traceback:
> line 48, in volmesh_from_polysurfaces, “C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas_rhino\geometry_constructors\volmesh.py”
> 

**To Reproduce**
Steps to reproduce the behavior:
1. Context [e.g. ST3, Rhino, Blender, ...]
2. Sample script
3. Sample data
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version [3.7]
 - Python package manager [conda]

**Additional context**
Add any other context about the problem here.

brgcode, 2020-04-07:
@juney-lee perhaps this can be addresses during the update of `compas_3gs`?

brgcode, 2020-04-14:
looking a bit closer at the error, the problem might already be solved by wrapping the result from `GetObject` in a list. the function returns a single guid, whereas the volmesh constructor expects a list of guids...

something like this might work

```python
guid = rs.GetObject(“select a closed polysurface”, filter=rs.filter.polysurface)
guids = [guid]
```

brgcode, 2020-04-14:
each polysurface in this list should be the b-rep of a cell in the volmesh. the volmesh constructor then uses the boundary loops of each of the polysurfaces to create the faces of the corresponding cell...

i think the data file `boxes.obj` is an example of correct input

brgcode, 2020-04-17:
is this working now?

brgcode, 2020-07-06:
@salmamzfr is this resolved?

salmamzfr, 2020-07-08:
Yes it is! Although I don't seem to find any documents on "volmesh_from_polysurfaces" on the website! I don't remember where I found this function originally!

salmamzfr, 2020-07-08:
Also after running this:

```
from compas_rhino.geometry._constructors import volmesh
import rhinoscriptsyntax as rs
guid = rs.GetObject(“select a closed polysurface”, filter=rs.filter.polysurface)
guids = [guid]
volmesh = volmesh.volmesh_from_polysurfaces(VolMesh, guids)
```

when I run 

`print volmesh`

I get this error: 

```
Message: 'VolMesh' object has no attribute 'edge'

Traceback:
  line 306, in data, "C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\volmesh\volmesh.py"
  line 216, in __str__, "C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\volmesh\volmesh.py"

```

brgcode, 2020-07-08:
but until the print statement everything is fine?

salmamzfr, 2020-07-08:
Yes!

brgcode, 2020-10-17:
@brgcode

brgcode, 2020-11-24:
volmesh has been completely restructured. this problem should no longer exist now...

closed 2020-11-24 by brgcode