compas-dev/compas #500: RPC Command-line interface quitting itself

yck011522, 2020-03-24:
**Describe the bug**
I sometimes face a situation where the RPC Proxy Server started from a command line terminates itself. No big deal, just great annoyance to have to manually restart it. Especially during development.

The context is that I'm using Proxy to call triangle library. From within Rhino. Identified reproducible code or steps, I have not. But I believe it is related to raised exception on the Server side functions. 

Does anyone faces some similar situation? #265 seems to reappear?
Or any idea why the RPC might kill itslef?

brgcode, 2020-04-07:
i noticed this as well with triangle.

it seems related to the fact that if there are small problems with the input data, triangle throws a segmentation fault which causes the RPC to shut down.

we resolved this in `compas_triangle` by properly cleaning the data before using it.

since then the problem has no longer occurred.

nothing more i can do about this as i cannot catch these kind of code errors to prevent the server from shutting down...

i will document the use of `compas_triangle` a bit better. perhaps you can give it a try and see if it helps.

brgcode, 2020-04-07:
examples for `compas_triangle` are here https://github.com/BlockResearchGroup/compas_triangle/tree/master/examples

the last one is specifically for rhino...

brgcode, 2020-04-08:
any last words?
without feedback, i will close this...

yck011522, 2020-04-09:
@brgcode 
Thanks for looking into this. I'm not aware of the internal implementation of the RPC. But I wonder if there is a easy way to wrap the actual function call (called by the RPC server) in some sort of try catch block? Such that RPC will handle exceptions (which occur rather often during development) gracefully.

brgcode, 2020-04-09:
that is already the case, but things like segmetation faults cannot be caught by that mechanism...

yck011522, 2020-04-09:
I see. Lets close the case then.

closed 2020-04-09 by yck011522