compas-dev/compas #503: Entrypoint compas_rpc seems to have hardcoded python path when installed from conda

tetov, 2020-03-31:
**Describe the bug**

Entrypoint `compas_rpc` (from [entrypoints in setup.py](https://github.com/compas-dev/compas/blob/master/setup.py#L77)) gives following output using conda installed compas:

```
(t2fofin) C:\Users\a\repos>compas_rpc start
Fatal error in launcher: Unable to create process using '"d:\bld\compas_1585547578076\_h_env\python.exe"  "C:\tools\miniconda3\envs\t2fofin\Scripts\compas_rpc.exe" start': The system cannot find the file specified.
```

**To Reproduce**
Steps to reproduce the behavior:
1. From anaconda env with compas 0.15.5
2. `compas_rpc start`

**Expected behavior**
This works fine in my `virtualenv` using editable pip installation.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python version 3.8
 - Python package manager conda

gonzalocasas, 2020-03-31:
Confirmed. This seems to be a bug recently introduced to `conda-build`: https://github.com/conda/conda-build/issues/3920

gonzalocasas, 2020-04-02:
@tetov according to [this](https://anaconda.org/conda-forge/compas/files), conda-forge has finished rebuilding this. Could give it another try and let us know if it works for you? If not, please re-open. (Before trying it out, run `conda clean -i` to make sure you have a refreshed index)

tetov, 2020-04-03:
@gonzalocasas I'm still getting same error. I tried with these builds:
```
compas-0.15.5-py38h32f6830_1
compas-0.15.5-py36h9f0ad1d_1
````
I've on conda 4.8.3 if that matters, miniconda installation.

I think you will have to re-open the issue, I don't see anything in the ui that would let me.

gonzalocasas, 2020-04-08:
Fixed, it was not a bug in conda-build, it was a problem in our feedstock (recipe) for `compas`.

closed 2020-04-08 by gonzalocasas