compas-dev/compas #504: Try integration tests using Rhino Compute / Rhino3dm

gonzalocasas, 2020-04-09:
In order to be able to run / test code paths that use `Rhino` SDK, it would be a possibility to try Rhino Compute / Rhino 3dm.

gonzalocasas, 2020-04-22:
To a certain extent, this can now be done with the test runner added on https://github.com/compas-dev/compas/pull/518

It does not allow to run code that really requires `Rhino` SDK thou, it only allows to run IronPython code paths and it fakes the presence of some Rhino stuff. But it would be a way to add integration tests based on this, and use `rhino.inside` to replace `Rhino` SDK.

jf---, 2024-04-06:
Compas has surpassed this in ambition 😉 -- `compas.rpc.Proxy` FTW

tomvanmele, 2024-04-08:
unfortunately not. we still don't have a mechanism to run automated tests as part of of our CI infrastructure for Rhino/Blender/...

jf---, 2024-04-08:
Gotcha, thanks, now I got the gist of @gonzalocasas proposal. Makes perfect sense.