compas-dev/compas #505: Network.from_nodes_and_edges does not work with nodes input as dictionary

salmamzfr, 2020-04-14:
**Describe the bug**
_Network.from_nodes_and_edges_ does not work with nodes input as dictionary! 
returns:

>     for i, (x, y, z) in enumerate(nodes):
> TypeError: cannot unpack non-iterable int object

**To Reproduce**
Steps to reproduce the behavior:
1. Context [e.g. ST3, Rhino, Blender, ...]
2. Sample script
3. Sample data
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version [e.g. 2.7]
 - Python package manager [e.g. macports, pip, conda]

**Additional context**
Add any other context about the problem here.

brgcode, 2020-04-14:
yes that is indeed missing.
have updated the code.
will release later today...

brgcode, 2020-04-14:
the master branch is already updated.
if you have a source install you could check it out and let me know if this solves your problem...

salmamzfr, 2020-04-14:
I used to have the compas-dev source folder on my project paths of Eclipse, so whenever I updated the source, I could run my codes with the updated source. Is it possible to do the same thing in Visual Studio Code?

brgcode, 2020-04-14:
if you are working from inside a conda environment, just install compas "from source" (using `pip install -e .`). wherever you activate this environment the checked out version of your source will then be available...

salmamzfr, 2020-04-14:
pip install -e .  encounters this error:

> Collecting planarity
>   Using cached planarity-0.4.1.tar.gz (164 kB)
>     ERROR: Command errored out with exit status 1:
>      command: 'C:\Users\msalma\AppData\Local\Continuum\anaconda3\envs\test\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\msalma\\AppData\\Local\\Temp\\pip-install-cgjonk0y\\planarity\\setup.py'"'"'; __file__='"'"'C:\\Users\\msalma\\AppData\\Local\\Temp\\pip-install-cgjonk0y\\planarity\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\msalma\AppData\Local\Temp\pip-install-cgjonk0y\planarity\pip-egg-info'
>          cwd: C:\Users\msalma\AppData\Local\Temp\pip-install-cgjonk0y\planarity\
>     Complete output (5 lines):
>     Traceback (most recent call last):
>       File "<string>", line 1, in <module>
>       File "C:\Users\msalma\AppData\Local\Temp\pip-install-cgjonk0y\planarity\setup.py", line 8, in <module>
>         from Cython.Distutils import build_ext
>     ModuleNotFoundError: No module named 'Cython'
>     ----------------------------------------
> ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

gonzalocasas, 2020-04-14:
I think that issue is documented [here](https://compas-dev.github.io/main/gettingstarted/knownissues.html).

salmamzfr, 2020-04-14:
When I install from the source with _pip install -e ._ and run my code in the same environment. I get an error for:  `from compas.datastructures import Mesh`

File "c:\users\msalma\documents\compas-dev\compas\src\compas\datastructures\__init__.py", line 178, in <module>
    from .mesh import *  # noqa: F401 F402 F403
  File "c:\users\msalma\documents\compas-dev\compas\src\compas\datastructures\mesh\__init__.py", line 8, in <module>
    from ._mesh import *  # noqa: F401 F403
  File "c:\users\msalma\documents\compas-dev\compas\src\compas\datastructures\mesh\_mesh.py", line 20, in <module>
    from compas.datastructures.mesh.transformations import mesh_transform
ImportError: bad magic number in 'compas.datastructures.mesh.transformations': b'\x03\xf3\r\n'

brgcode, 2020-04-14:
are you still using compas from eclipse as well? do you have a different Python version there?

salmamzfr, 2020-04-14:
No, I don't think so! I am not using eclipse anymore! I had an old version of anaconda that I just uninstalled today.  The python version on the current environment is: Python 3.6.10 I have other environments with python 2 and other COMPAS versions, but this environment I just made today to test the pip install.

brgcode, 2020-04-14:
hmm weird. this procedure should work quite reliably because i (and many others) use it regularly on both mac and windows. do you have a working from source install in one of the Python 2 environments?

salmamzfr, 2020-04-15:
I had it on my base, and just removed it, unfortunately the problem persists! :( 
I even removed the test environment with the source compas and created a new one, still didn't help!

brgcode, 2020-04-15:
putting stuff in your base is a bad idea. if you don't have too many environments, try removing them one by one and reset your base environment with `conda install --revision 1`. then leave the base environment alone and install COMPAS in a new environment.

brgcode, 2020-04-17:
everything working now?

salmamzfr, 2020-04-17:
I have not reset my base environment yet, since I have several environments that I am constantly working with. I will do it on the weekend. Is a conda version with the correction for the "Network.from_nodes_and_edges" already out? I just updated compas to version. 0.15.5, but still get the same error.

salmamzfr, 2020-04-20:
While trying to reset the base, I get the following error: 

> PackagesNotFoundError: The following packages are missing from the target environment:
>   - anaconda::conda==4.7.12=py37_0
>   - anaconda::openssl==1.1.1c=he774522_1

Do you know any cure for that?

brgcode, 2020-04-20:
not really, would have to google around...
(you could try `conda clean --all`)

btw, are you trying to reset the base while keeping all the other environments? not sure if that is at all possible.

sometimes, after too much fiddling, it is easier to just reinstall Anaconda and start fresh.
might be a lot quicker as well...

salmamzfr, 2020-04-20:
I have already removed all the envs. I was also thinking to remove Anaconda and reinstall it!

salmamzfr, 2020-04-20:
Well, I uninstalled and reinstalled Anaconda, and made one environment for the compas source with 
_pip install -e ._ (I have not made any other environments yet!). Still get this error for importing Mesh:     

> from compas.datastructures import Mesh
>   File "c:\users\msalma\documents\compas-dev\compas\src\compas\datastructures\__init__.py", line 178, in <module>
>     from .mesh import *  # noqa: F401 F402 [...]

salmamzfr, 2020-04-20:
I removed the compas source folder from my computer and cloned it again from github and recreated the environment. Finally "Network.from_nodes_and_edges" works! :D

brgcode, 2020-04-20:
👍 

can i close this then? :)

salmamzfr, 2020-04-20:
Sure! :)

closed 2020-04-20 by brgcode