compas-dev/compas #510: Numba 0.49 dropping Python 2 support.

tetov, 2020-04-17:
[Numba 0.49.0](https://numba.pydata.org/numba-doc/latest/release-notes.html#version-0-49-0-apr-16-2020) dropped Python 2 support. This showed up in a [travis job when pip tried to build llvmlite 0.32 from sdist](https://travis-ci.com/github/tetov/compas_fab/jobs/320507740#L315) (https://github.com/compas-dev/compas_fab/pull/138) . Previous versions of llvmlite provided wheels for py27.

I think pinning Numba to `numba<0.49` should be enough.

brgcode, 2020-04-17:
@gonzalocasas how does stuff like this relate to the transpiling? making our code backwards compatible is one thing, but we can't keep missing out on updates because of it...

tetov, 2020-04-17:
Or `numba <0.49; python_version >= '2.7' and python_version < '3.0'`.

Functions requiring numba are already marked with `_numba` right? And they didn't work from Rhino before. So if Rhino is the only reason to keep supporting Python 2, dropping numba support for Python 2 doesn't change much.

We could make Python 3 a build/install requirement, but still support IPY from Rhino/Grasshopper.

brgcode, 2020-04-17:
yes, you are right. the Python 3 build will indeed still use the latest versions.

tetov, 2020-04-23:
Fixed in #518?

gonzalocasas, 2020-04-23:
I would say so

closed 2020-04-23 by tetov