compas-dev/compas #512: How to get a compas Mesh from grasshopper

yck011522, 2020-04-20:
**Describe the bug**
Maybe this is a simple question.

In the old days, there is a function "mesh_from_guid" in the helpers. I notice it is gone now.
Is there a new function / suggested syntax to convert a GH created Mesh (Not referenced Mesh from Rhino), It is a Rhino.Geometry.Mesh object into a compas_mesh?

yck011522, 2020-04-20:
Ok, I found a function in a file @Matteo gave me a while ago that can do what I wanted:
```
def rMesh2cMesh(rg_mesh):
    if isinstance(rg_mesh, rg.Mesh):
        # single mesh
        compas_mesh_vertices = rg_mesh.Vertices.ToPoint3dArray()
        compas_mesh_faces = rg_mesh.Faces.ToIntArray(True)
        compas_mesh_faces = [compas_mesh_faces[i:i + 3] for i in range(0, len(compas_mesh_faces), 3)]
        return Mesh.from_vertices_and_faces(compas_mesh_vertices, compas_mesh_faces)
    else:
        # array of meshes
        meshes = []
        for m in rg_mesh:
            compas_mesh_vertices = m.Vertices.ToPoint3dArray()
            compas_mesh_faces = m.Faces.ToIntArray(True)
            compas_mesh_faces = [compas_mesh_faces[i:i + 3] for i in range(0, len(compas_mesh_faces), 3)]
            meshes.append(Mesh.from_vertices_and_faces(compas_mesh_vertices, compas_mesh_faces))
        return meshes
```
Maybe someone knows better, if there is a function already like this in compas.
@romanarust 
Feel free to close the issue.

brgcode, 2020-04-20:
`RhinoMesh.from_guid(guid).to_compas()`

brgcode, 2020-04-20:
there is still a small bug in this function as pointed out on the forum but it will disappear soon...

brgcode, 2020-04-20:
if the above does not work there is also 
`mesh = RhinoMesh.from_geometry(geometry).to_compas()`

and the method `to_compas` has an optional `cls` parameter in case you want something else than `compas.datastructures.Mesh` in return...

brgcode, 2020-04-20:
the bug is discussed here
https://forum.compas-framework.org/t/compas-v-0-15-5-unknown-error/331

romanarust, 2020-04-21:
as mentioned by @brgcode, `RhinoMesh.from_geometry(geometry)` should do the job for GH.
See https://github.com/compas-dev/compas/pull/501

yck011522, 2020-04-22:
Hi @brgcode @romanarust 
Got it. Thank you very much. Both works well. 
I'm not entire sure if geometry created in GH (never baked to Rhino) has GUIDs as well. I observed the following behavior:

GH Python component - Input Field- Type Hint - **Set to Mesh**
- print(type(mesh_input)) -> <type 'Mesh'>
- [Works] RhinoMesh.from_geometry(mesh_input)
- [Error] RhinoMesh.from_guid(mesh_input)

GH Python component - Input Field- Type Hint - **Set to (ghdoc Object when Geometry)**
- print(type(mesh_input)) -> <type 'Guid'>
- [Works] RhinoMesh.from_geometry(mesh_input)
- [Works] RhinoMesh.from_guid(mesh_input)

The error message is not surprising. I'm just going to leave this observation here for future.
Runtime error (ArgumentTypeException): expected Guid, got Mesh

closed 2020-04-22 by yck011522