compas-dev/compas #513: Adding clickable link to base classes methods in online documentation

yck011522, 2020-04-20:
# Feature Request

- Add docs for base classes methods such as those in BaseNetwork, Datastructure. 
- Allow their methods to be seen in the docs of their inherited class docs. Such as in Network.
- In the docs, where there is the list of class methods / functions, label those that comes from base class. For example in some library documentation like [this](https://docs.microsoft.com/en-us/dotnet/api/system.windows.media.drawing?view=netframework-4.8), there is a clickable label saying (Inherited from BaseClassName).

## Details

**Describe the solution you'd like**

Today, I'm trying to inherit the newly reworked 0.15.x  Network class. I read the docs/tutorial about the Network class implementation and I checked out all the class methods it has. There are access methods for `edge_attribute` and `node_attributes` but not network_attributes. 

Took me a long time to dig out where and how I can use that. Of course, I remember I opened #482 a while ago. But this Feature Request is more about making the online doc more navigable.

brgcode, 2020-04-21:
you are right. the docs are a bit behind and do not reflect the latest changes. 
working on it now. will push updates regularly throughout the week...

yck011522, 2020-04-21:
Cool. By the way, I also noticed that in the old days, we have a clickable [source] link from the docs. And that is very useful. Now those are gone. 

Perhaps this is related to some restructuring of where the functions are located and they might be in different files. But if it is easy to bring the links back, please do. They are very convenient.

brgcode, 2020-04-21:
yes this broke during one of the intermediate updates but will come back indeed.

gonzalocasas, 2022-03-28:
Is this done? Can it be closed?

jf---, 2024-04-08:
@gonzalocasas anytime you close two unicorns are united in an eternal bond

tomvanmele, 2024-04-08:
this is generally solved but there are a few instances where one of the classes in the chain is not part of the api docs. will be completely resolved very soon. considering the unicorns' eternal bliss, i think we can close it...