compas-dev/compas #520: Add headers to all files written by COMPAS

brgcode, 2020-04-23:
* COMPAS version
* file type
* file info
* author
* date modified
* ???

gonzalocasas, 2020-04-23:
Here's a proposal:

* Formalize and describe in the documentation the `data` formats used by COMPAS (first task would be to enumerate them all)
* Introduce schemas for each of them. We can define schemas in whatever format we feel like, the obvious option being [json schema](https://json-schema.org/), but others could be considered (flatbuffers, protobuf, thrift, cbor/cddl)
* Newly defined schemas should be assigned version numbers and documented.
* Add unit tests that verify conformance to the schemas.

The header of the schemas should be compatible across all of them, I would propose the following fields:
 * Schema type (equivalent to file type, this would be equivalent to the 'signature' or 'magic numbers' in some binary file formats, we should assign it a unique enough value)
 * Schema version (schema evolution should not be tied to compas version implementing it)
 * Generator (software and version that generated the file. In the case of COMPAS, this would be, eg `COMPAS 0.15.5`.)
 * System Version (ie output of `sys.version`, eg `2.7.8 (IronPython 2.7.8 (2.7.8.0) on .NET 4.0.30319.42000 (64-bit))`)
 * Author (Sure, why not)

I don't think dates are needed, since that is metadata that is carried on the file itself, but if you see a reason to add that, then why not. Perhaps `content-length` would also be useful if the thing is to be sent over sockets.

brgcode, 2020-04-24:
json schema for all file formats? or just for the "data dicts" of COMPAS objects?

gonzalocasas, 2020-04-24:
just for the "data dicts" of COMPAS objects

gonzalocasas, 2020-04-24:
(since we're describing a header format, I thought the question was limited to our own formats/data dicts, not to external formats)

brgcode, 2020-04-24:
i hadn't thought of the data dicts, but good point, that should indeed happen as well!

about the external formats, though.
i find it a bit annoying that the files in the wild seem to be very unreliable in terms of their implementation of the formats and thought we could at least improve this for the ones we write...

gonzalocasas, 2020-04-24:
So, external formats means `obj`, `ply`, `off`,  and `gltf`, right? (ie, those we currently support).

Maybe I'm missing something, but I is tricky to add headers to those formats. Either they already have a header format (and likely one that doesn't fit our purpose), or have none and no way to add one except as comments (some don't even have comments).

Where do you see the proposed header fitting? As comment(s)?

brgcode, 2020-04-24:
yes that is how it did it so far.

have a look here https://github.com/compas-dev/compas/blob/5930604b8719f5fab5891d22924e8e11e43da87e/src/compas/files/obj.py#L404

but it was just an idea, perhaps also to be able to trace back errors in the file writing and so on. it doesn't interfere with the format of the file itself...

gonzalocasas, 2020-04-24:
In that case, I would go for something minimal, like a single-line comment with `"COMPAS: {}, PYTHON: {}".format(compas.__version__, sys.version)` , to explicitly limit it to aid in debugging / tracing back issues.

<sub>
According to <a href="https://xkcd.com/927/">ancient wisdom</a>, we should avoid the idea of building "a better standard for the existing standard" 😉
</sub)

brgcode, 2020-04-24:
let's split this up into two problems and link one of them to #525 

not because of the `jsonpickle` thing but because better use of the encode/decoder stuff together with proper schema/header info will make this kind of problems obsolete...

gonzalocasas, 2020-08-24:
Should we close this already  or still needs work?

brgcode, 2020-08-24:
we can close this. improvements are always possible but those would deserve their own ticket...

closed 2020-08-24 by brgcode