compas-dev/compas #522: Mesh.delete_face(fkey) raises KeyError when used on triangulated Rhinocommon meshes.

tetov, 2020-04-23:
**Describe the bug**
`Mesh.delete_face(fkey)` raises `KeyError` when used on triangulated Rhinocommon meshes.

`Mesh.is_quadmesh()` returns `True` for some reason, and return from `Mesh.insert_vertex(fkey)` contains an extra face consisting of vertex keys `[x, x, y]` instead of expected `[x, y, z]`.

**To Reproduce**
Steps to reproduce the behavior:
1. Grasshopper
2. Sample script
```
from compas_rhino.geometry import RhinoMesh
from compas_ghpython.artists import MeshArtist

mesh = RhinoMesh.from_geometry(mesh).to_compas()

assert mesh.is_quadmesh()

key = mesh.get_any_face()
mesh.delete_face(key)

artist = MeshArtist(mesh)
rhino_mesh = artist.draw_mesh()
```
[image]
4. See error
```
Runtime error (KeyNotFoundException): KeyError

Traceback:
  line 737, in delete_face, "C:\Users\a\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\mesh\core\halfedge.py"
  line 10, in script
```

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Rhino 6 SR4 2.7.8 (IronPython 2.7.8 (2.7.8.0) on .NET 4.0.30319.42000 (64-bit))
 - `compas==0.15.5` installed using conda.

brgcode, 2020-04-23:
i guess this is related to the fact that all mesh faces in rhino are "quads" in the sense that triangle faces also have 4 vertices but with vertices 3 and 4 being the same

tetov, 2020-04-24:
Is this something that should we checked when using one of `compas_rhino.geometry.RhinoMesh`'s classmethods? Or is it better to do in `to_compas()`?

So check for and remove duplicate vertex keys in faces (by checking lengths of sets of face vkey or a RhinoCommon face attribute) somewhere in the `RhinoMesh.from_geometry(mesh).to_compas()` flow?

brgcode, 2020-04-24:
yes it has already been discussed here https://forum.compas-framework.org/t/compas-v-0-15-5-unknown-error/331
and will be fixed later today...

brgcode, 2020-04-24:
the fix has been pushed and is building.
please check if it fixes your problem...

closed 2020-04-25 by tetov

tetov, 2020-04-25:
Fixed in 857a29e0