compas-dev/compas #523: RobotModel.add_joint() limit parameter (upper,lower) is inconsistent with documentation.

yck011522, 2020-04-23:
**Describe the bug**

Refering to the `RobotModel.add_joint()` function, the limit parameter.

https://github.com/compas-dev/compas/blob/f710bd42c6e510cc23482a68c5c0cd9f3e240ef6/src/compas/robots/model/robot.py#L770-L823

Documentation says:

> limit : list of 2 float
>             The lower and upper limits of the joint 

However, in the code, there is this line where the upper goes before the lower limit:

```
if limit:
   upper, lower = limit
   limit = Limit(lower=lower, upper=upper)
```

gonzalocasas, 2020-04-23:
Agree.
@romanarust which way is correct? Should we correct the code to be `lower, upper= limit` or is the documentation incorrect?

romanarust, 2020-04-23:
ups! yes, should be `lower, upper`. line 821 needs to be changed

commit 759a9a21 (2020-04-28): fix #523
closed 2020-04-28 by romanarust