compas-dev/compas #525: compas.geometry.Point cannot be pickled

yck011522, 2020-04-24:
**Describe the bug**
I'm using jsonpickle to serialize data. It works well with Mesh but not Point (Frame also doesn't work).

**To Reproduce**

install json pickle and run the following script.
```
import jsonpickle
from compas.geometry import Point
from compas.datastructures import Mesh

jsonpickle.set_preferred_backend('json')
jsonpickle.set_encoder_options('json', indent=4)

# Json Pickle works like a charm on Mesh
mesh = Mesh.from_polyhedron(6)
print ((jsonpickle.encode(mesh)))

# Json Pickle does not work on Point
x = Point(1,2,3)
print ((jsonpickle.encode(x)))
```

Output for mesh (This is good):
```
{
    "attributes": {
        "name": "Mesh"
    }, 
    "facedata": {
        "5": {}, 
        "4": {}, 
        "1": {}, 
        "0": {}, 
        "3": {}, 
        "2": {}
    }, 
    "py/object": "compas.datastructures.Mesh", 
    "edgedata": {}, 
    "default_edge_attributes": {}, 
    "_max_int_key": 7, 
    "face": {
        "5": [
            4, 
            5, 
            6, 
            7
        ], 
        "4": [
            4, 
            7, 
            1, 
            2
        ], 
        "1": [
            0, 
            1, 
            7, 
            6
        ], 
        "0": [
            0, 
            3, 
            2, 
            1
        ], 
        "3": [
            4, 
            2, 
            3, 
            5
        ], 
        "2": [
            0, 
            6, 
            5, 
            3
        ]
    }, 
    "default_vertex_attributes": {
        "z": 0.0, 
        "y": 0.0, 
        "x": 0.0
    }, 
    "vertex": {
        "5": {
            "z": 1.1547005383792517, 
            "y": 1.1547005383792517, 
            "x": -1.1547005383792517
        }, 
        "4": {
            "z": 1.1547005383792517, 
            "y": 1.1547005383792517, 
            "x": 1.1547005383792517
        }, 
        "7": {
            "z": 1.1547005383792517, 
            "y": -1.1547005383792517, 
            "x": 1.1547005383792517
        }, 
        "6": {
            "z": 1.1547005383792517, 
            "y": -1.1547005383792517, 
            "x": -1.1547005383792517
        }, 
        "1": {
            "z": -1.1547005383792517, 
            "y": -1.1547005383792517, 
            "x": 1.1547005383792517
        }, 
        "0": {
            "z": -1.1547005383792517, 
            "y": -1.1547005383792517, 
            "x": -1.1547005383792517
        }, 
        "3": {
            "z": -1.1547005383792517, 
            "y": 1.1547005383792517, 
            "x": -1.1547005383792517
        }, 
        "2": {
            "z": -1.1547005383792517, 
            "y": 1.1547005383792517, 
            "x": 1.1547005383792517
        }
    }, 
    "_max_int_fkey": 5, 
    "halfedge": {
        "5": {
            "4": 3, 
            "6": 5, 
            "3": 2
        }, 
        "4": {
            "5": 5, 
            "7": 4, 
            "2": 3
        }, 
        "7": {
            "4": 5, 
            "6": 1, 
            "1": 4
        }, 
        "6": {
            "7": 5, 
            "0": 1, 
            "5": 2
        }, 
        "1": {
            "7": 1, 
            "0": 0, 
            "2": 4
        }, 
        "0": {
            "6": 2, 
            "1": 1, 
            "3": 0
        }, 
        "3": {
            "5": 3, 
            "0": 2, 
            "2": 0
        }, 
        "2": {
            "4": 4, 
            "1": 0, 
            "3": 3
        }
    }, 
    "default_face_attributes": {}
}
```
Output for Point (Not good, it is empty):
```
{
    "py/object": "compas.geometry.Point"
}

```
**Expected behavior**
Expected the second output to contain the point data

**Desktop (please complete the following information):**
 - Python version: Tried both in RhinoGHPython and cPython 3.6.7
- compas version 0.15.5

brgcode, 2020-04-24:
only difference that could cause this that i can think of is that the geometry objects use slots...

yck011522, 2020-04-24:
@brgcode  Interesting. I can look into it.

yck011522, 2020-04-24:
@brgcode Sadly, that is not the reason.

```import jsonpickle
from compas.geometry import Point

class MyPoint(object):
    __slots__ = ['_x', '_y', '_z']
    def __init__(self, x, y, z):
        self._x = x
        self._y = y
        self._z = z

# Json Pickle does not work on compas Point
x = Point(1,2,3)
print ((jsonpickle.encode(x, warn = True, indent = 4, keys = True)))

# Json Pickle does work on MyPoint
y = MyPoint(1,2,3)
print ((jsonpickle.encode(y, warn = True, indent = 4, keys = True)))

```
Prints:
```
{
    "py/object": "compas.geometry.Point"
}
{
    "_x": 1, 
    "py/object": "__main__.MyPoint", 
    "_y": 2, 
    "_z": 3
}
```

yck011522, 2020-04-24:
I'm looking at the __init__() of the Point class, looks like there are both __slot__ assignment and __dict__ assignment. I'm not sure how does this work. From what I read online, it seems you can only do one of those.

https://github.com/compas-dev/compas/blob/e0ea5e344fad158707791eae2ff2f61a65e6a68d/src/compas/geometry/_primitives/point.py#L112

In fact if I change my code to this, it will throw no attribute error:
```
class MyPoint(object):
    __slots__ = ['_x', '_y', '_z']
    
    def __init__(self, x, y, z):
        self._x = 0.0
        self._y = 0.0
        self._z = 0.0
        self.x = x
        self.y = y
        self.z = z
```

I also compared `Point` and `MyPoint` like this:
```
# Json Pickle does not work on Point
compas_point = Point(1,2,3)
print ((jsonpickle.encode(compas_point, warn = True, indent = 4, keys = True)))

my_point = MyPoint(1,2,3)
print ((jsonpickle.encode(my_point, warn = True, indent = 4, keys = True)))

print(compas_point.__slots__)
print(compas_point.__dict__)

print(my_point.__slots__)
print(my_point.__dict__) # This throws no attribute '__dict__' error
```
It looks like the problem is the compas point have both _slots__ and __dict__. And jsonpickle got confused.

brgcode, 2020-04-24:
the `self.x` and so on are not dict assignments. they are descriptors (properties) assigning values to the slots...

brgcode, 2020-04-24:
can you provide a use case for what you are trying to accomplish? perhaps i can provide a snippet of how i would do this with the functionality that is already there...

brgcode, 2020-04-24:
just as a follow up.
you cannot have classes with slots and normal attributes. that would throw an error. the dict you are referring to is empty, which is guess is why jsonpickle has no problems with it...

```python
>>> p = Point(0, 0, 0)
>>> p.__dict__
{}
```

yck011522, 2020-04-24:
I think I'm pretty sure the issue is related to the fact compas point has a dictionary. It will break jsonpickle even if the dictionary is empty. In the following code, I instantiated the __dict__ forcefully and this cause the jsonpickle not encode.

```
import jsonpickle

class MyBrokenPoint(object):
    __slots__ = ['_x', '_y', '_z', '__dict__']
    
    def __init__(self, x, y, z):
        self._x = 0.0
        self._y = 0.0
        self._z = 0.0

my_point = MyBrokenPoint(1,2,3)

print ((jsonpickle.encode(my_point, warn = True, indent = 4, keys = True)))
print(my_point.__slots__)
print(my_point.__dict__) 
```

Output

```
{
    "py/object": "__main__.MyBrokenPoint"
}

['_x', '_y', '_z', '__dict__']

{}
```

yck011522, 2020-04-24:
What I want to do:

I am writing a package with my own classes, some of them inheriting from Network, RobotModel etc. Some of these classes stores compas.geometry objects and I would like my class to be serializable in a human-readable format. In particular, 
- I want them to recursively serialize. 
- I do not want to write serialization handlers for each classes. 

jsonpickle does both tasks auto-magically. At least up until I store compas.geometry in them.

brgcode, 2020-04-24:
```python
from compas.datastructures import Datastructure

from compas.geometry import Point
from compas.geometry import Vector

class Victor(Datastructure):

    def __init__(self, points=None, vectors=None, mixed=None):
        self.points = points
        self.vectors = vectors
        self.mixed = mixed

    def to_data(self):
        data = {'points': self.points, 'vectors': self.vectors, 'mixed': self.mixed}
        return data

    @classmethod
    def from_data(cls, data):
        victor = cls()
        victor.points = data['points']
        victor.vectors = data['vectors']
        victor.points = data['mixed']
        return victor

if __name__ == '__main__':

    import json
    from compas.utilities import DataEncoder
    from compas.utilities import DataDecoder

    points = [Point(0, 0, 0) for i in range(2)]
    vectors = [Vector(0, 0, 0) for i in range(2)]
    mixed = [Point(0, 0, 0)] + [Vector(0, 0, 0)]

    victor = Victor(points, vectors, mixed)

    data = json.dumps(victor, cls=DataEncoder)

    print(data)

    victor = json.loads(data, cls=DataDecoder)

    print(victor.points)
    print(victor.vectors)
    print(victor.mixed)

```

yck011522, 2020-04-24:
@brgcode 
I found a 1 line solution to fix this problem. See if you think this is worth pulling.

If we add a empty `__slots__ = []` decoration in the Primitive class, then the Point class can be serialized. This line suppress the creation of the empty __dict__ in the Primitive class and thus will not appear in the child classes. 

The Primitive class will look like this: 
```
class Primitive(object):
    """Base class for geometric primitives."""

    __module__ = "compas.geometry"
    
    __slots__ = []

    def __init__(self):
        pass

    ...
    ...
```

I think this doesn't

brgcode, 2020-04-24:
okay will have a look

brgcode, 2020-04-24:
how about you submit a PR then we can discuss together with @gonzalocasas 

we were going to work on the serialisation mechanisms anyway...

brgcode, 2020-04-24:
have a look at the solution above as well though

yck011522, 2020-04-25:
@brgcode As a follow up discussion to your example. I appreciate this way of implementing a custom serialization handler `to_data()` and `from_data()`, in fact I was using this method for a long time until I discovered jsonpickle.

The reason is that from my development perspective, where I'm the only developer and probably there will only be one user in the coming year or so, to maintain all the serialization handlers for all my classes is a lot of work. Especially when I'm deciding on what class attributes I should have for each class and they are changing rapidly. Therefore a quick recursive serialization saves a lot of work.

I understand from a compas framework development perspective, it will be important to have custom serialization handlers for data integrity conforming to an invariant schema, like what was discussed in #520, but also for security (avoid json code injection in pickled files). This is however not important and very time consuming for a small package like mine.

For (a long and boring) example I have `TimberBeam(object)` and `TimberJoint(object)` which have Points and Frames stored. I have `TimberAssembly(Network)` and I store many `TimberBeam` and `TimberJoint` in the `Network.node_attributes` and `Network.edge_attributes`. I have then a `ClampingProcessManager(object)` that stores this `TimberAssembly`, and also some classes that describe robots and tools that are inherited from `compas.RobotModel`. It is just way easier if all I have to do is:
```
import jsonpickle
f = open(some_location)
f.write(jsonpickle.encode(myobject))
```

PR #526 (2020-04-27): Addig __slot__ = [] to _primitive
Fixed #525.
The __slot__ = [] declaration in geometry.Primitive suppress the creation of __dict__ in all the child class such as Points, Frame etc... 

This is motivated by the need for these classes to be serializes correctly by jsonpickle.

### What type of change is this?

- [x] Bug fix in a **backwards-compatible** manner.
- [ ] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that  [... cut]
closed 2020-04-27 by brgcode