compas-dev/compas #531: How to get compass Vector from grasshopper

yck011522, 2020-04-28:
# Feature Request

Related to #512 
I noticed there is no `RhinoVector` wrapper. Is there a way to get vector input from GH interface and convert it to `compas.geometry.Vector`

I would like to do something like this:
```
vector = RhinoVector.from_geometry(geometry).to_compas()
```

yck011522, 2020-04-28:
@romanarust

yck011522, 2020-04-28:
And also 1 more question: 
How do I convert a `compas.geometry.Vector` back to a Rhino Vector that I can output via the ghpython component.

tetov, 2020-04-28:
> 
> 
> And also 1 more question:
> How do I convert a `compas.geometry.Vector` back to a Rhino Vector that I can output via the ghpython component.

v = Vector([0, 0, 0])
a = rg.Vector3d(*v.data)

yck011522, 2020-04-28:
@tetov Thanks for the PR I look forward to it.

gonzalocasas, 2020-04-28:
you don't need to use `v.data`, just cast it to a list before you pass it to the constructor of `rg.Vector3d`:

```python
import Rhino.Geometry as rg
from compas.geometry import Vector

rhino_v = rg.Vector3d(100, 20, 10)
print (rhino_v)

compas_v = Vector(*rhino_v)
print (compas_v)

rhino_v2 = rg.Vector3d(*list(compas_v))
print (rhino_v2)
```

brgcode, 2020-04-28:
can we close this with the merged PR?

yck011522, 2020-04-29:
Looks good to me.

closed 2020-04-29 by yck011522