compas-dev/compas #538: Calculation of geodesic distances starting from many vertices

ioanna21, 2020-05-04:
**Describe the bug**
I would like to report a problem I have with the heat geodesics calculation (compas.datastructures.mesh_geodesic_distances_numpy). When I calculate geodesic distances with one vertex as a start point, the result is correct. However, when I calculate with multiple start points, then the result is wrong. (see screenshots below)

**To Reproduce**
I have created this simple example that illustrates the problem. You can download it here:
https://www.dropbox.com/sh/qg10kywmiugh52c/AADV7J0VQkB9hDmgoZ-pEZeza?dl=0
Run the code to see the distance gradient that is produced starting from multiple vertices.

**Expected behavior**
The correct gradient starting from all these vertices should be parallel to the boundary line.  As a comparison, I also attach images of the geodesic distances (again using the same heat algorithm), calculated in libigl, with the exact same inputs.  

**Screenshots**

**Compas**

Geodesic distances starting from one vertex. These seem to be correct. (starting vertex in blue)

[image]

Geodesic distances starting multiple vertices. These are wrong.  (starting vertices in blue)

[image]

**Libigl**
Geodesic distances starting from one vertex. (starting vertex in red)

[image]

Geodesic distances starting multiple vertices. (starting vertices in red)

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version: 3.7.6
 - Python package manager: conda-forge

brgcode, 2020-05-04:
will have a look. have you tried increasing the value of `m`?

ioanna21, 2020-05-04:
Many thanks for the reply! Indeed you are right. I had not tried using such a high value for m, because in the heat geodesics paper m=1 is recommended. They say that higher values would create a smooth approximation of the geodesic distances.  But maybe this m=1000 just accounts for the different units? I will test it with m=1000 on more complex shapes and will let you know if I have more problems.

brgcode, 2020-05-04:
in any case the exact method and the heat method of `libigl` are available through `compas_libigl`, which by the end of this week should be conda installable...

ioanna21, 2020-05-04:
Great, thanks for the info! On the same example, with m = 1000 and a some better coloring, I still see some problems in the resulting distances. 

[image]

closed 2020-05-05 by ioanna21