compas-dev/compas #542: Mesh.join(mesh)

yck011522, 2020-05-08:
# Feature Request

Can we have a function to join two Mesh.

## Details

I would like to combine two mesh (may or may not overlap) into one single Mesh object
Where:
new_mesh.vertices=mesh1.vertices+ mesh2.vertices (these Index are renumbered)
new_mesh.faces = mesh1.faces + **mesh2.faces (mapping to new indices)

This is more of a basic function and there are many uses. Perhaps should be implemented at Graph level, but I'm not sure.

**Additional context**
This is similar to the join function in Rhino.

brgcode, 2020-05-09:
that function exists but i noticed it is not in the docs...
`compas.datastructures.meshes_join`

yck011522, 2020-05-09:
Arh. Nice. Maybe add it to the docs?

yck011522, 2020-10-28:
Fixed by #583

closed 2020-10-28 by yck011522