compas-dev/compas #543: Transformation `from_frame()` should be named `to_frame()`

yck011522, 2020-05-10:
# Feature Request

This is more of a naming/terminology discussion instead of a feature request.

From a user perspective, "Transformation" are typically thought of to transform an object, say from A into B. 

In that sense, if I would want to transform something from a frame A to frame B. I would get that `Transformation` by calling function `Transformation.from_frame_to_frame(frame_from, frame_to)`. This makes sense to me grammatically. Particularly the word "from" and "to" is expressed from the perspective of the object that is being transformed.

However, I think the function that is currently named `Transformation.from_frame(frame)` is not consistent with this line of thought. This function, is essentially the same as `Transformation.from_frame_to_frame(Frame.worldXY(), frame)`  but allows us to skip the `frame_from` parameter. In my opinion, this should be named `Transformation.to_frame`, and the signature should be `Transformation.to_frame(frame_to)`.

## Details

**Describe the solution you'd like**
Renaming function `Transformation.from_frame(frame)` to `Transformation.to_frame(frame_to)`

**Additional context**
In my opinion, `Frame.to_world_coords(object_in_lcs)` and `to_local_coords(object_in_wcf)` is also consistent with the meaning the word "to" expressed from the perspective of the object.

romanarust, 2020-05-11:
Good point! 
The prefix `from` tells you that the function a class constructor. All `Class.from_*` are  class constructors in compas. So you actually have to read it in that way: `Transformation.from_(frame)`  and `Transformation.from_(frame_to_frame)`.  
So, renaming it to `to_frame` would help for clarification locally, but it would create inconsistency. Another option would be renaming `Transformation.from_frame` to
`Transformation.from_worldXY_to_frame`.  It doesn't look so nice, though. Inputs there?

yck011522, 2020-05-11:
I understand the word `from` from the POV of the constructor. But in this case, the class is a geometrical action where the word `from` carry very descriptive meaning. 

In this line of argument, the function `Transformation.from_frame(frame)`  (if it shall continue to exist)   should actually perform `Transformation.from_frame_to_frame(frame, Frame.worldXY())` and `Transformation.to_frame(frame)` should perform `Transformation.from_frame_to_frame(Frame.worldXY(), frame)` 

(A few thoughts later) Okay, but the `to_frame` maybe more troubling because that might sound like a cast from Transformation to a Frame..... 

Perhaps the name `Transformation.from_frame_source()` and `Transformation.from_frame_target()` for the two behavior I list above.

brgcode, 2020-10-19:
can i close this?

yck011522, 2020-10-19:
@brgcode I'm not aware if this is resolved in some way or another, but if there is no further discussion interest, we can close this.

closed 2020-11-24 by brgcode